micropython/micropython #3213: can set global variable inside interrupt but can't test agents it?

kenthinson, 2017-07-15:
Hello. Sorry if this is obvious but I'm completely new to python and micro python.

here is my issue. if I run 
`pin = Pin(4, Pin.IN)
firstFire = 0

def pressedButton(p):
        firstFire = 1
        print(firstFire)
pin.irq(trigger=Pin.IRQ_FALLING | Pin.IRQ_RISING, handler=pressedButton)`

this seems to work as I can print out the firstFire value and it's 1

however if I try instead 
`pin = Pin(4, Pin.IN)
firstFire = 0

def pressedButton(p):
    if firstFire == 0:
        firstFire = 1

pin.irq(trigger=Pin.IRQ_FALLING | Pin.IRQ_RISING, handler=pressedButton)`

I get the error NameError: local variable referenced before assignment. Is this a api issue or am I just doing something totally wrong? thank you.

robert-hh, 2017-07-15:
Try pressedButton() as:
```
def pressedButton(p):
    global firstFire
    if firstFire == 0:
        firstFire = 1
```

kenthinson, 2017-07-16:
thanks! works great

closed 2017-07-16 by kenthinson