micropython/micropython #3217: missing sys.{excepthook,displayhook}

pmp-p, 2017-07-17:
Hi, some use cases i could not complete because did not find micropython equivalent for those on unix-android/ esp8266
sys.excepthook:
     restore terminal sane settings on failure when using picotui.
     reset boards to *safe* state on emergency.
     remote debbuger helper ( transmit filename / line / msg to ide )

sys.displayhook:
    allow infinite repl enhancements. ( xonsh shell like, pretty print etc ... )
    and simple stdout multiplexing/filtering hacks when using coroutining or extra displays.

dpgeorge, 2017-07-17:
sys.displayhook is available in uPy via the (non-standard) built-in `__repl_print__`.  You can call it, and assign a new 1-arg function to it to change it.

pmp-p, 2019-07-06:
@dpgeorge is there something against renaming ```__repl_print__``` to ```sys.displayhook``` ?

i'd like to try to add sys.excepthook in order to make an async repl like i did for cpython https://github.com/pmp-p/aioprompt.
as a step to solve a specific problem of javascript port https://github.com/pmp-p/micropython-ports-wasm/issues/3

dpgeorge, 2019-07-09:
> is there something against renaming `__repl_print__` to `sys.displayhook` ?

It does seem like a good idea.  But it's not currently possibly to modify elements of a built-in module like `sys`, so that's the main blocker.