micropython/micropython #3218: RFC sensor API: updates?

hoihu, 2017-07-17:
sorry for not appending to #2093 but I'm unable to comment there (Gihub shows "You can't comment at this time") - for whatever reason that is.

I would like to update my [SenseHAT](https://www.raspberrypi.org/products/sense-hat/) drivers to the machine API. I'm currently working on the first one (HTS221, temp&humidity sensor) and started to implemented some functionality as described by @dpgeorge. 

However I'm unsure if the driver API is stable...RFC #2093 has been silent for a while now and I haven't seen any driver popping up in the forum or elsewhere that follow the proposed guidelines.

So a couple of questions to hopefully revive this discussion:
- Is the sensor API now fixed as per propsal made by @dpgeorge?
- Are there "reference" drivers available that fulfill the proposal so they can be used as a guideline for driver developers? 
- Do we need to agree on a set of get_xx accessor methods for `SensorBase`? Are they going to be provided by the main repository?

If interested, I can provide a draft version of my WIP driver that I think comes close the the API (as a discussion base).

hoihu, 2017-08-07:
I've updated this issue with an example, see #3256

jonnor, 2024-07-02:
I believe this issue can be closed, as the MR has been closed.

projectgus, 2024-07-05:
Definitely close this in favour of the original RFC issue now, I think. Hopefully that issue can be commented on, again!

closed 2024-07-05 by projectgus