micropython/micropython #3220: [ESP8266] RTC discrepacy between doc and firmware

PikWay, 2017-07-18:
https://docs.micropython.org/en/latest/esp8266/library/machine.RTC.html

My version:
MicroPython v1.9.1-167-gcadbd7f on 2017-07-18; ESP module with ESP8266

>>> import machine
>>> rtc=machine.RTC()
>>> rtc.init((2017, 7, 18, 23, 05, 0, 0, 0))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'RTC' object has no attribute 'init'

Probbably documentation issue. 

-Adam

candeira, 2017-08-14:
The RTC implementation for esp8266 is different enough from that of pyboard (and also from the API declared at https://github.com/micropython/micropython/wiki/Hardware-API#the-rtc-real-time-clock-class):
```
>>> help(RTC)
object <class 'RTC'> is of type type
  datetime -- <function>
  memory -- <function>
  alarm -- <function>
  alarm_left -- <function>
  irq -- <function>
  ALARM0 -- 0
```

The implementation is not even a subset of the RTC HW API, as there is one extra getter/setter method: `memory()`, which accesses the non-volatile memory. There is one issue pointing to documentation of the RTC HW in the ESP8266 here: https://github.com/micropython/micropython/pull/2158, specifically in this commit: https://github.com/micropython/micropython/pull/2158/commits/5af924f06bf130639f7d59f360a48ff1bd58b968. The commit wasn't accepted, but the info in it could be reused in any documentation rewrite.

My open question would be why is the function call `memory()` part of the RTC module, and not part of the hardware-specific `esp` API hierarchy.

There are some other existing issues and comments already dealing with the API/documentation mismatch:
https://github.com/micropython/micropython/issues/1430#issuecomment-253200785 and https://github.com/micropython/micropython/issues/1430#issuecomment-254418124 which discuss the API side, and https://github.com/micropython/micropython/pull/2334, which discusses the docs side.

nubcore, 2018-01-29:
I am updating the docs for the esp8266 implementation, not sure how to generate a pull request that will only apply to the esp8266 docs and not other ports?

peterhinch, 2018-01-29:
If you look at `docs/library/machine.Timer.rst` you'll find some syntax which restricts output to individual ports.

dpgeorge, 2019-12-17:
In 04e7aa056316d7b4400b508caefe26c6b75384f8 the docs were updated to mention which RTC methods are missing on esp8266.

closed 2019-12-17 by dpgeorge