micropython/micropython #3223: Unix: initialisation of multi-line dict can report wrong line number for syntax error.

peterhinch, 2017-07-19:
The following:
```python
a = {
    'one' : 1,
    'two' : 2,
    'three' : 3,
    'four', 4  # error
    }
```
produces the following outcome:
```
MicroPython v1.8.7-893-g46b849a on 2017-06-22; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import rats35
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "rats35.py", line 2, in <module>
SyntaxError: expecting key:value for dict
>>> 
```

pfalcon, 2017-07-19:
This is more or less clear: the line no where dict definition starts is reported. I'm sure this is won't fix. If you're concerned, you may want to add diffcase similar to http://docs.micropython.org/en/latest/pyboard/genrst/builtin_types.html?highlight=line%20number#exception-in-while-loop-condition-may-have-unexpected-line-number

dpgeorge, 2017-07-21:
It's possible to improve the situation slightly: the compiler currently uses the head element of the dict/set to report the line number for the error, but it can in fact use the actual element that it's up to, which in some cases would lead to the correct line number given for the error.  But not in the example given here because `four` and `4` are leaf nodes in the parser and don't carry any location (file, line) information.

Applying the patch to make it try to report the element it's up to would in the case above change the error for the line number from "line 2" to "line 1" because that's where the opening brace of the dict is.  This is arguably a better error because it's not pointing to the first element and potentially confusing the user.

The patch is very small and just uses one variable instead of another, but makes bare-arm increase by 4 bytes, probably because the C compiler changes its register allocation.