micropython/micropython #3227: File size of files larger than 2 GB is shown as negative number

robert-hh, 2017-07-22:
All versions:
For files larger than 2 GB the file size is displayed as negative number, resulting from a int <-> uint mismatch in the code for uos.stat() (function fat_vfs_stat() in vfs_stat.c [which I added myself]). The error is in line 246. Does anyone know the proper macro to pick up the fno.fsize value as unsigned int?
Besides the showing value, it's nothing to care about.

pfalcon, 2017-07-22:
> The error is in line 246

Please provide URL linking to this line on github.

robert-hh, 2017-07-22:
Here is is:
https://github.com/micropython/micropython/blob/master/extmod/vfs_fat.c#L246

dpgeorge, 2017-08-09:
Thanks @robert-hh for the report.  It's not a straightforward fix because:
1. At first thought one could use mp_obj_new_int_from_uint(), since this will return a small-int object if the number fits, otherise a bignum.
2. But different ports have different small-int sizes (eg 15 bit on 16-bit MCU, or 63-bit on 64-bit machine), and the argument to mp_obj_new_int_from_uint() is of type mp_uint_t, so one may lose higher bits if passing a large file offset to this function.
3. One could use mp_obj_new_int_from_ull(), whose argument on all archs/ports is large enough to cover a 64-bit file size.  But that function always constructs a bignum even if the arg will fit in a small int, and this will make the most common cases of uos.stat() use more RAM.

So a comprehensive fix needs a new int constructor function that takes an unsigned-long-long and returns a small int if possible, otherwise a bignum.

pfalcon, 2017-08-09:
> At first thought one could use mp_obj_new_int_from_uint(), since this will return a small-int object if the number fits, otherise a bignum.

I guess this should be good enough initial workaround, allowing to deal with up to 4GB files on a 32-bit platform. (I also guess initial @robert-hh's description is a bit incorrect, as files above 1GB would have negative size; extending supported range 4x is pretty good for starters.)

dpgeorge, 2017-08-21:
This was fixed (at least converted to using mp_obj_new_int_from_uint) for extmod/vfs_fat and unix/modos in 4c736ea8fc046dc564f9167967a5dd92f07ed002

closed 2017-08-21 by dpgeorge