micropython/micropython #3231: [PyBoard 1.1] Suspect UART race

dbc, 2017-07-24:
Don't have a test case yet, but wanted to capture the issue. 

Version info: Micropython 1.9,1, PyBoard v1.1

With two different devices on two different UARTs, I am seeing crashes bad enough to wipe the file system when the UART is crap flooded. I'm debugging new drivers, so at times the uart input buffer is getting filled while the crap flood continues. Occasionally, the board spontaneously resets, some of those crashes wipe the file system.  My suspicion is a race condition that occurs when the input buffer is full.  

Probably can replicate by using an FTDI cable set for 115200 or there about, and sending a continuous stream, with a slow-ish consumer process on the PyBoard that lets the buffer fill.

dpgeorge, 2017-09-08:
@dbc did you make any further progress with this, are you able to provide a test case?

jonnor, 2024-09-19:
No information provided for reproducing the problem.

closed 2024-09-19 by jonnor