micropython/micropython #3232: const() fails on 1.9.1

dbc, 2017-07-24:
micropyton/test/micropython/const.py fails on 1.9.1

    >>> from micropython import const
    >>> X = const(123)
    >>> Y = const(X+456)
    Traceback (most recent call last):
      File "<stdin>", line 1
    SyntaxError: constant must be an integer

dpgeorge, 2017-07-24:
Please specify your uPy port.  The test works on the unix version for me.

peterhinch, 2017-07-24:
Fails here on Unix version built last Friday.
```
MicroPython v1.8.7-1023-g6ede921 on 2017-07-21; linux version
```

dpgeorge, 2017-07-24:
The test works because it's executed in a single file.  The const() feature does not work at the REPL because the parser/compiler does not save constants from one invocation to the next (but X and Y are still in the global namespace).

dbc, 2017-07-24:
PyBoard v1.1. I was testing at the REPL, including copying/pasting from the tests directory in repo.  But that was *after* my module threw errors on import.  I can post the code later today when I have access to it again.

dbc, 2017-07-25:
Hmmm... OK so on a PyBoard 1.0, with  pybv10-20170611-v1.9.1.dfu the test works if imported from the REPL. 

My previous testing was on a PyBoard 1.1, which looking at my workspace, I suspect I flashed to pybv11-network-20170611-v1.9.1.dfu -- why I was using network I don't know, I think my mouse speared the wrong pixel when I downloaded. I don't want to muck with that board tonight since I just carefully ran through a dress-rehearsal for a demo.  Rather than tempt fate, I will defer further testing by a couple of days.

dbc, 2017-08-05:
I can verify that this is a REPL-only issue for me.  When I filed the issue I was going back and forth between REPL and import, so I wasn't sure any more if I was seeing the bug only in REPL or if I also saw it on import.

dbc, 2017-08-05:
This fails:

```python
from micropython import const
import struct

s = '>III'
n = struct.calcsize(s)
print (n)
print (type(n))
SIZE = const(n)
```
SyntaxError: constant must be an integer

Should this work?  
SIZE=const(struct.calcsize(s)) also fails.

Comment out SIZE= to see that n is <class 'int'>

peterhinch, 2017-08-06:
I think the arg to const() has to be known at compile time. The result of executing arbitrary Python code is unknown to the compiler.

curtisdave, 2017-08-07:
@peterhinch I see the point.  I guess I'm used to languages where certain functions, such as string length, are evaluable at compile time in order to enable constant folding. struct.calcsize(), of course, is evaluated at import time -- a different beastie.

I wonder if it might be worth making a Micropython-specific extension to struct to add a calsizeconst() method?  Or perhaps the idiom is not common enough to be worth the effort.

Perhaps there is a work-around idiom for Micropython where BUFF_FMT and PKT_LEN=struct.calcsize() are delegated to a module, so that then PKT_LEN is bound at import time and therefore is available to const() in the importing module.

peterhinch, 2017-08-08:
There are two purposes for `const()`. One is to save RAM, the other is as an optimisation for the compiler. If you declare a const as a global in a module this will not save RAM, as the module might be imported. To declare a const in a module such that it saves RAM, the instance name must be prepended with `_` so that it is unavailable for import. So the `PKT_LEN` will necessarily use RAM.

The compiler optimisation is to avoid a lookup:
```python
_FOO = const(42)
_BAR = const(_FOO * 7)

def my_func(a):
    return a ** 2

z = my_func(_BAR):
    # code omitted
```
The compiler will call `my_func()` with its argument computed at compile time, saving an indirection.

When compiling module "A" the compiler won't compile and execute imported module "B" to compute and extract compile-time objects. After all, the compiler won't complain if module "B" does not exist at compile time; the error is at runtime. A constant declared in a module (without an underscore) and imported into another will use RAM and, in the importing module, will be accessed at runtime so the indirection will still be present.
Note that the last point is AIUI and I am happy to be corrected.

pfalcon, 2017-09-16:
Summing up:

* An argument of const() should be compile(parse)-time constant expression (of course).
* Constants are module-local and don't work across modules (as constants, across modules they're just variables).
* Each REPL clause is evaluated as a standalone module.  Following works:

~~~
x=const(1); y=const(x + 1)
~~~

methoxid, 2017-10-04:
This example in the docs does not work then, at least in REPL ,
any recommendation on how to use named variables in @micropython.asm_thumb functions? 

v1.9.2-125-gc9a0b2a-dirty on 2017-10-03

https://docs.micropython.org/en/latest/pyboard/reference/asm_thumb2_hints_tips.html#named-constants

```
MYDATA = const(33)

@micropython.asm_thumb
def foo():
    mov(r0, MYDATA)
```

dpgeorge, 2017-10-04:
@methoxid if you use paste mode at the REPL (ctrl-E, paste the code, ctrl-D) then that example will work.

dpgeorge, 2022-06-23:
`const` works as designed, to help define constant symbols and data local to a .py file.

closed 2022-06-23 by dpgeorge