micropython/micropython #3233: assert len(pad) >= 4 AssertionError in "makeimg.py" when building firmware-combined.bin

ashareye, 2017-07-24:
hi
        I write a simple softuart function to mp, just like uart0, using xtensa-lx106-elf-gcc-4.8.5, it shows:
> LINK build/firmware.elf
   text	   data	    bss	    dec	    hex	filename
 578444	   1080	  64648	 644172	  9d44c	build/firmware.elf
Create build/firmware-combined.bin
esptool.py v1.2
('flash    ', 37072)
Traceback (most recent call last):
  File "makeimg.py", line 25, in <module>
    assert len(pad) >= 4
AssertionError
make: *** [build/firmware-combined.bin] Error 1
make: *** Deleting file `build/firmware-combined.bin'

         Is the size of bin file more than 4M? I only wrote about 300 lines. please tell me what's happen?        thanks

dpgeorge, 2017-07-24:
The esp8266 has a very limited amount of iRAM and you need to put most (all) of your code in to flash.  To do that you need to add your new .c source files to the list in esp8266_common.ld, see https://github.com/micropython/micropython/blob/master/esp8266/esp8266_common.ld#L76

ashareye, 2017-07-24:
Thank you very much for your reply!  It works for me.

closed 2017-07-24 by ashareye