micropython/micropython #3237: py: Weird overflow during calculation

hoihu, 2017-07-24:
Consider:
```
def fixed_to_float(fx):  return (fx >> 8) * 10**((fx & 0xff) - 128)

p1 = ((250 * 3964519 // 1024) << 8) | (-3 +128) 
p2 = ((1000 * 3964519 // 4096) << 8) | (-3 +128) 

print("p1={} of type {}".format(p1,type(p1)))
print("p2={} of type {}".format(p2,type(p2)))
print(fixed_to_float(p1))
print(fixed_to_float(p2))
```
which yields
```
p1=247782525 of type <class 'int'>
p2=247782525 of type <class 'int'>
967.9
0
```

using the exact same input values (p1=p2) of the same type, the function yields 2 different results (!).

The same code runs fine in CPython 3.5.2

I'm on:
```
MicroPython v1.9.1-153-gad3abcd3 on 2017-07-09; PYBv1.0 with STM32F405RG
```

danicampora, 2017-07-24:
I'm running 
```
MicroPython v1.8.6-715-g5fa16c2c on 2017-07-20; LoPy with ESP32
```

And I can reproduce it as well. Weird indeed...

dpgeorge, 2017-07-25:
Nice find!  And thanks for the report.  A smaller test that exhibits the same behaviour (on unix uPy) is:
```python
big = 1 << 70
base = 10

print(base ** (-1 + big - big))
print(base ** (-1))
```

dpgeorge, 2017-07-25:
Fixed in 04552ff71b6c722b21597d93481f024c72457cef

closed 2017-07-25 by dpgeorge