micropython/micropython #3241: uasyncio emits no error message if a coro is called rather than awaited.

peterhinch, 2017-07-25:
The following
```python
import uasyncio as asyncio

async def bar():
    await asyncio.sleep(1)
    print('In bar')

async def foo():
    print('Launching bar')
    await bar()
    print('Getting it wrong')
    bar()  # bad

loop = asyncio.get_event_loop()
loop.run_until_complete(foo())
```
produces this output
```
>>> import rats37
Launching bar
In bar
Getting it wrong
>>> 
```
CPython 3.5.2 issues the following error message:
```
>>> import rats37
Launching bar
In bar
Getting it wrong
/home/adminpete/rats37.py:11: RuntimeWarning: coroutine 'bar' was never awaited
  bar()
>>> 
```
It would be useful if a message were emitted in response to this error.

hiway, 2017-07-25:
I too recently stumbled into realizing this the long/hard way. An error message will be very useful.

tralamazza, 2017-08-09:
Just curious, are coro supposed/expected to be reentrant?

peterhinch, 2017-08-10:
This can't arise because (u)asyncio is a cooperative scheduler. Reentrancy can only occur in preemptive scheduling. It is possible to have multiple instances of the same coro, but instance `X` will only be scheduled when instance `Y` has yielded control. Instance `Y` will be re-scheduled when `X` has yielded. In other words there is no re-entrancy.

tralamazza, 2017-10-27:
Probably I don't know enough of mpy, so things cannot be called directly from an interrupt context?

peterhinch, 2017-10-28:
This type of question is best raised in the [forum](https://forum.micropython.org/).
An ISR is a function or bound method rather than a coroutine, so it cannot `await` a coroutine. The only way it can schedule a coroutine for execution is with
```python
def my_isr():
    loop = asyncio.get_event_loop()
    loop.create_task(foo())
```
If this ISR executes at a high rate multiple copies of `foo` will be scheduled for execution, which will occur when that copy reaches the top of the event queue. An individual copy of `foo()` will never re-enter. This is one of the (many) benefits of cooperative over pre-emptive scheduling - complications such as reentrancy simply can't occur.

For some interesting general reading on the distinction between synchronous and asynchronous functions I suggest reading [this](http://journal.stuffwithstuff.com/2015/02/01/what-color-is-your-function/). There is an unofficial tutorial on uasyncio [here](https://github.com/peterhinch/micropython-async.git) - follow the link to "tutorial".

pfalcon, 2017-10-28:
Back to the original report:

@peterhinch : Do you know how that's implemented in CPython? That will never be implemented in uasyncio. The whole idea of uasyncio was to stay reasonably compatible with asyncio (provide CPython-compatible wrappers where it isn't, e.g. https://github.com/micropython/micropython-lib/tree/master/cpython-uasyncio), so development and debugging could happen with it and all the debugging bloat it offers.

Granted, that never was picked up by community, and looks less viable as MicroPython gets more and more optimized APIs which CPython doesn't have.

There's an option of (trying to) adding such an option on VM level, then have a "testing" MicroPython build to develop/test apps (existing micropython_coverage build can be reused for that for starters).

Another option is to maintain patched version of uasyncio with debugging features (but that right away would work only for legacy decorated coros, etc.).

peterhinch, 2017-10-29:
Given that it adds significant complexity perhaps the easily-made error is best caught by a sourcecode parsing utility - if I'm made a monkey by it again I'll write one :)

Hopefully pylint will pick this up in due course.

tralamazza, 2017-10-30:
@peterhinch thank you for taking the time to answer. I meant calling the coro directly, as in your example 😄  (not using await). In any case I took a look at the async code, it's all good.

peterhinch, 2017-10-30:
@tralamazza Calling it directly is a coding error. I haven't performed a detailed study of what happens under MicroPython when you do it but my impression - supported by the example in my root post - is that the code silently fails to run. My interest is in identifying lines containing this error so it can be fixed rather than studying its consequences; for what it's worth re-entrancy won't occur if the code never actually runs!

peterhinch, 2017-12-12:
I have written a [script](https://github.com/peterhinch/micropython-async/blob/master/check_async_code.py) which aims to locate this error in otherwise syntactically correct source files. It is rather crude and can occasionally produce false positives but it's saved me a lot of head-scratching when code silently fails to work as expected.

Improvements always welcome.

peterhinch, 2018-07-01:
[This PR](https://github.com/micropython/micropython-lib/pull/292) produces an assertion failure (and hence a traceback) if a script issues `create_task` or `run_until_complete` with a generator function instead of a generator/coro.

If (in error) `await` is issued with a generator function, a traceback already occurs.