micropython/micropython #3242: uasyncio StreamReader/StreamWriter can fail if a coro blocks on an awaitable

peterhinch, 2017-07-25:
The following runs on a Pyboard with a link between X1 and X2. If the `Bar` class is changed such that the sleep time is zero, the UART I/O does not take place.
```python
# Link X1 and X2 to test.

import uasyncio as asyncio
from pyb import UART
uart = UART(4, 9600)

async def sender():
    swriter = asyncio.StreamWriter(uart, {})
    while True:
        await swriter.awrite('Hello uart\n')
        await asyncio.sleep(2)

async def receiver():
    sreader = asyncio.StreamReader(uart)
    while True:
        res = await sreader.readline()
        print('Recieved', res)

class Bar():
    def __await__(self):
        while True:
            yield from asyncio.sleep_ms(5)  # Change this to 0 and the app fails

    __iter__ = __await__

bar = Bar()

async def foo():
    print('Starting foo. Should never finish.')
    await bar
    print('foo finished.')

loop = asyncio.get_event_loop()
loop.create_task(foo())
loop.create_task(sender())
loop.create_task(receiver())
loop.run_forever()
```

peterhinch, 2017-07-27:
Here is a more minimal test case:
```python
import uasyncio as asyncio
from pyb import UART
uart = UART(4, 9600)

async def sender():
    swriter = asyncio.StreamWriter(uart, {})
    while True:
        await swriter.awrite('Hello uart\n')
        await asyncio.sleep(2)

async def receiver():
    sreader = asyncio.StreamReader(uart)
    while True:
        res = await sreader.readline()
        print('Recieved', res)

async def foo():
    while 1:
        yield 0  # UART only works if > 0

loop = asyncio.get_event_loop()
loop.create_task(foo())
loop.create_task(sender())
loop.create_task(receiver())
loop.run_forever()
```

peterhinch, 2017-07-27:
The following fix works with ~25% hit on scheduling performance but its performance relies on a behaviour of moduselect.c which differs from CPython.
core.py:
```python
    def run_forever(self):
        cur_task = [0, 0, 0]
        while True:
            if self.q:
                if time.ticks_diff(self.q.peektime(), self.time()) <= 0:
                    self.wait(0)  # Ensure IO gets scheduled.
                # wait() may finish prematurely due to I/O completion,
                # and schedule new, earlier than before tasks to run.
                while 1:
```
The problem is that `wait(0)` calls `self.poller.ipoll(delay, 1)` with delay 0. According to CPython docs this causes blocking but under MicroPython it effects a zero delay. A call of `wait(1)` is compatible but inevitably wrecks performance with its 1ms delay.

I'm happy to submit a PR but I'd appreciate comments on these points:
 1. Is there a better approach?
 2. Can the behaviour of `.ipoll(0, 1)` be assumed to be as at present?

jpochyla, 2017-07-27:
In [our case](https://github.com/trezor/trezor-core/blob/master/src/trezor/loop.py#L106-L110) we decided to always give higher priority to tasks blocked on I/O.

peterhinch, 2017-07-30:
A way to minimise the performance hit in cases where the IORead/IOWrite mechanism is not in use would be for the `PollEventLoop` to maintain a count of registered objects in `.reg_objects`:
```python
    def run_forever(self):
        cur_task = [0, 0, 0]
        while True:
            if self.q:
                if self.reg_objects and time.ticks_diff(self.q.peektime(), self.time()) <= 0:
                    self.wait(0)  # Ensure IO gets scheduled.
                # wait() may finish prematurely due to I/O completion,
                # and schedule new, earlier than before tasks to run.
                while 1:
```
But I'd appreciate an answer to 2. above before preparing a PR.

dpgeorge, 2017-07-31:
> Can the behaviour of .ipoll(0, 1) be assumed to remain as at present?

Delays of 0 are a pure poll and never wait.  This is true also for CPython for both select.select and select.poll; see https://docs.python.org/3/library/select.html#select.select (" A time-out value of zero specifies a poll and never blocks.") and https://docs.python.org/3/library/select.html#select.poll.poll ("If timeout is omitted, negative, or None, the call will block until there is an event for this poll object.")

So, yes, ipoll with a zero timeout should always have the behaviour of a zero delay.

peterhinch, 2017-07-31:
This PR uses this approach https://github.com/micropython/micropython-lib/pull/201.

pfalcon, 2017-08-20:
Based on the title, this should have been submitted to micropython-lib project. Further discussion is by the ticket linked above.

closed 2017-08-20 by pfalcon