micropython/micropython #3243: softuart with baudrate=115200, the first byte is always wrong

ashareye, 2017-07-26:
hi
  
 I have referred to [here](https://github.com/plieningerweb/esp8266-software-uart) and made some changes to build a software uart for esp8266 and had to set baudrate 115200,  since my peripheral is at that rate.

 But the strange thing is that the first word is always wrong, whether sending or receiving, then the subsequent bytes are correct. I adjusted the parameters and made the connection test using Arduino, cp2102, PL2303 etc, but got the same result. 

I suspect that someting is wrong with the first timing of set and read gpio, just like stm32, please give some advise, thank you.  

here is the  [my code](https://github.com/ashareye/softuart.git)

dpgeorge, 2017-07-26:
In general we can't support 3rd party libraries like softuart, and a better place to discuss this is at https://forum.micropython.org .

The thing that comes to mind is that the timing critical code is running from flash and the first time round it must be loaded to the cache, so that throws off the timing for the first byte.  To try and fix it, put Softart_Intr_Handler and Softuart_Putchar in iRAM.

ashareye, 2017-07-26:
Thank you very mach for your reply, since the limitation of iRAM, i put the softuart.o into flash, and how can i put the two functions in iRAM only? to new a sepcial file that only contains two functions and do not comment it in flash in esp8266_common.ld?

dpgeorge, 2017-07-26:
> how can i put the two functions in iRAM only? to new a sepcial file that only contains two functions and do not comment it in flash in esp8266_common.ld?

Yes that's one way to do it.  You can also use the MP_FASTCODE macro (grep for existing use of this to learn how).

ashareye, 2017-07-26:
thanks again, i  will try it

ashareye, 2017-07-26:
aha, it works! thank you very much!

closed 2017-07-26 by ashareye