micropython/micropython #3246: cpp-coveralls came and broke Travis build

pfalcon, 2017-07-28:
Last unfailed build:

https://travis-ci.org/micropython/micropython/builds/257565772

~~~
Successfully installed cpp-coveralls-0.3.12
~~~

First failing build:

https://travis-ci.org/micropython/micropython/builds/258107900

~~~
Successfully installed cpp-coveralls-0.4.0
...
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 863, in resolve
    new_requirements = dist.requires(req.extras)[::-1]
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2577, in requires
    "%s has no such extra feature %r" % (self, ext)
pkg_resources.UnknownExtra: urllib3 1.7.1 has no such extra feature 'secure'
The command "(cd unix && coveralls --root .. --build-root . --gcov $(which gcov) --gcov-options '\-o build-coverage/' --include py --include extmod)" exited with 1.
~~~

Proposed solution: force-install cpp-coveralls==0.3.12

pfalcon, 2017-07-28:
> Proposed solution: force-install cpp-coveralls==0.3.12

I implemented this workaround in f578947ae3fee5610c5bc1123baf878b92eaa248, because I'd like to perform refactors and have CI test them. Keeping this open if someone wants to dig deeper.

dpgeorge, 2017-09-08:
The fix is discussed in https://github.com/eddyxu/cpp-coveralls/issues/118 and #3306 provides a working implementation.

dpgeorge, 2017-09-10:
Fixed by #3306.

closed 2017-09-10 by dpgeorge