micropython/micropython #3247: Inconsistent typecasts in .locals_dict initialization: void* vs mp_obj_dict_t*

pfalcon, 2017-07-28:
999cedb90f alone has both forms. Let's decide what should be the correct one, and apply it consistently, otherwise current situation leads to confusion and copy-paste errors. (I for one thought that correct is (void*)).

pfalcon, 2017-07-29:
In https://github.com/micropython/micropython/commit/f2140f94466f043677d58352bbecbdbe98e82d20 I'm using (mp_obj_dict_t*) as a more specific cast.

dpgeorge, 2017-07-31:
void* will work but mp_obj_dict_t* is more specific, so let's use that (and I think the cast is only necessary to cast away the const-ness, if the mp_obj_t struct had `const mp_obj_dict_t* locals_dict` as the entry then the pointer wouldn't need any casting).

pfalcon, 2017-08-09:
Applied as suggested.

closed 2017-08-09 by pfalcon