micropython/micropython #3250: Question: Prepoluate file system on an ESP8266

binloan, 2017-07-31:
Hello, this is less an issue but a support request maybe also for some additional documentation.

I'm currently building my custom esp firmware with some extra modules (because of heap issues). I would like to make it as simple as possible later to run my software so I thought that it should be possible to add the main.py (or any other additional file) already inside the firmware. So that later the user has only a single step to get my software running, instead of four or five.

I see that the _boot.py is somewhat executed and copied into the file system. The question for me now is: How can I get this done for other files? Thank you already for any help--

peterhinch, 2017-07-31:
This type of query is best raised in the forum; see my suggestion in response to a similar query [there](https://forum.micropython.org/viewtopic.php?f=16&t=3612&p=20971&hilit=_boot.py#p20971).

binloan, 2017-07-31:
I have the impression it doesn't really answer my question. I will continue however as suggested.

closed 2017-07-31 by binloan