micropython/micropython #3251: HID on ESP8266 please?

idesignstuff, 2017-08-01:
How hard is it to port the PYB HID module for ESP8266 boards?

I'm using Wemos D1 Mini, and Micropython has been fun to start learning on.

I'd really like to be able to plug my Wemos into a Win/Mac/Linux usb port, and send keystrokes and/or mouse inputs via HID. Mass storage would be nice as well.
 Then the wifi can be where I go to configure what the functions will be for each physical button wired up to the Wemos board...

peterhinch, 2017-08-02:
This is really a topic for the [forum](https://forum.micropython.org/) rather than the bug tracker. Mass storage can be done via the official `sdcard.py` module (in the source tree under drivers/sdcard).

idesignstuff, 2017-08-03:
Ok thanks.

mattytrentini, 2017-08-14:
As discussed in the [HID library/module for ESP8266](https://forum.micropython.org/viewtopic.php?f=16&t=3640) forum post, porting the PyBoard's HID to the ESP8266 isn't possible without extra hardware (or, while _technically possible_ it's not a great idea!).  

Suggest closing this issue?

idesignstuff, 2017-08-15:
Yes. Fair enough.

Christian

closed 2017-08-15 by idesignstuff

mattytrentini, 2017-08-15:
Thanks @idesignstuff.