micropython/micropython #3255: Automatic (re)building of mpy-cross negatively affects development cycle

pfalcon, 2017-08-05:
Ever since mpy-cross started to build automatically on "make" (e.g. on unix) port, there's additional (2x or more) wait time on each change to rebuild it. This makes development more cumbersome. There should be a special make target to build just local upy binary, without updating mpy-cross, or some other workaround.

dpgeorge, 2017-08-16:
> or some other workaround.

The issue is not the mpy-cross dependency but rather the frozen bytecode dependency.  And it's possible to remove this dependency (and hence not build mpy-cross0:
```
$ make FROZEN_MPY_DIR=
```
That will (or at least should) work for all ports.  Though it may require a fresh build the first time, since it changes mpconfig.h options.

pfalcon, 2017-08-17:
> The issue is not the mpy-cross dependency but rather the frozen bytecode dependency.

I'm not sure I agree. I'm fine with bytecode dependency, and would like to build executable as configured in mainline (e.g. with upip frozen for unix port). What I wouldn't like is mpy-cross rebuilt on every sneeze, as probability that something affecting mpy-cross changed in current working session is low. Is there provision to ignore frozen bytecode's dependency on mpy-cross (i.e. make it all work like before, by passing something on the make command line) ?

SpotlightKid, 2017-11-29:
Any progress on this? Halving the typical compile time is worthwhile goal IMO.

dpgeorge, 2019-06-03:
Resolved by 41739506589ec8397613c86d8f682fb7f86c0a9f

closed 2019-06-03 by dpgeorge