micropython/micropython #3257: ESP8266: Btree module fails on creating or using a database with more than a few entries

robert-hh, 2017-08-08:
This issue was brought up by @crizeo in the forum. Creating (and using) a btree database fails, if once it gets a reasonable size. The following code fails on an ESP8266:
```
import btree
import os, urandom  # for random bytestring + db size
import gc  # for showing free heap
from ubinascii import hexlify
DATABASE_FILENAME = "btree.db"

def randbytes():
    length = 1 + urandom.getrandbits(3)  # 1 to 16
    return os.urandom(length)

try:
    dbfile = open(DATABASE_FILENAME, 'r+b')
except OSError:
    dbfile = open(DATABASE_FILENAME, 'w+b')
db = btree.open(dbfile, cachesize=2560)

counter_entries = 0  # number of entries (pairs of k+v) in db
counter_bytes = 0  # number of bytes (keys + values) in database
for i in range(500):
    key, value = "%d" % counter_entries, hexlify(randbytes())
    db[key] = value  # here is where the error occurs
    counter_entries += 1
    counter_bytes += len(key) + len(value)
    if counter_entries % 10 == 0:
        print("total: %d entries (%d B); db size: %d B; free heap: %d B" %
              (counter_entries, counter_bytes, os.stat(DATABASE_FILENAME)[6], gc.mem_free()))

    gc.collect()

db.close()
dbfile.close()
```
The log output is:
```
>>> import test_btree
total: 10 entries (104 B); db size: 0 B; free heap: 12576 B
total: 20 entries (206 B); db size: 0 B; free heap: 12528 B
total: 30 entries (344 B); db size: 0 B; free heap: 12544 B
total: 40 entries (476 B); db size: 0 B; free heap: 12544 B
total: 50 entries (592 B); db size: 0 B; free heap: 12544 B
total: 60 entries (698 B); db size: 0 B; free heap: 12544 B
total: 70 entries (808 B); db size: 0 B; free heap: 12544 B
total: 80 entries (914 B); db size: 0 B; free heap: 12544 B
total: 90 entries (982 B); db size: 0 B; free heap: 12544 B
total: 100 entries (1124 B); db size: 0 B; free heap: 12528 B
total: 110 entries (1258 B); db size: 0 B; free heap: 12544 B
total: 120 entries (1392 B); db size: 0 B; free heap: 12544 B
total: 130 entries (1548 B); db size: 0 B; free heap: 12560 B
total: 140 entries (1644 B); db size: 0 B; free heap: 12544 B
total: 150 entries (1762 B); db size: 0 B; free heap: 12544 B
total: 160 entries (1886 B); db size: 0 B; free heap: 12544 B
total: 170 entries (2018 B); db size: 0 B; free heap: 4272 B
total: 180 entries (2130 B); db size: 0 B; free heap: 4288 B
total: 190 entries (2264 B); db size: 0 B; free heap: 4272 B
total: 200 entries (2400 B); db size: 0 B; free heap: 4272 B
total: 210 entries (2514 B); db size: 0 B; free heap: 4288 B
total: 220 entries (2646 B); db size: 0 B; free heap: 4288 B
total: 230 entries (2794 B); db size: 0 B; free heap: 4272 B
total: 240 entries (2936 B); db size: 0 B; free heap: 4288 B
total: 250 entries (3050 B); db size: 0 B; free heap: 4288 B
total: 260 entries (3164 B); db size: 0 B; free heap: 4288 B
total: 270 entries (3290 B); db size: 0 B; free heap: 4272 B
total: 280 entries (3420 B); db size: 0 B; free heap: 4288 B
total: 290 entries (3542 B); db size: 0 B; free heap: 4288 B
total: 300 entries (3674 B); db size: 0 B; free heap: 4272 B
total: 310 entries (3790 B); db size: 0 B; free heap: 4272 B
total: 320 entries (3916 B); db size: 0 B; free heap: 4288 B
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "test_btree.py", line 24, in <module>
OSError: 0
```
The same code runs fine on Linux Micropython with a constant value for free heap. If the database is created on a different machine and copied to the ESP8266, it fails too.

(sysname='esp8266', nodename='esp8266', release='2.0.0(5a875ba)', version='v1.9.1-208-g4dc7c56-dirty on 2017-08-07', machine='ESP module with ESP8266')

robert-hh, 2017-08-08:
Fixed: One has to set the pagesize in the call of btree.open() to a small value, like 1024.

pfalcon, 2017-08-09:
Let me reopen this, I'd like to look what happens there when I have time. Likely, it's out of memory issue with large pages cached in memory.

robert-hh, 2017-08-09:
From the log above it looks as if the default pagesize is 8k, and since at least 5 pages are requires, according to the comments in btree.h, that fails. With a pagesize of 512, my test of creating a database with 500000 entries on an ESP8266 succeeded, even if it took like 20 hours. I guess that a small pagesize limits the size of a key/value pair to a combined 256 byte.
Setting the default per port looks easy, since in extmod/btree.c, in the code for btree.open() the actual default is set to 0. That could also be a define from mpconfigport.h.

robert-hh, 2017-08-10:
I ran the same test script on ESP32 with the esp32 port. There is not problem with the default values, and more interesting almost no change in available free heap space (96 byte max-min. That differs a lot from the behavior on ESP8266, where even with a  pagesize of 512, with a difference of  ~1760 bytes.
So it seems that the ports differ on where the buffers are taken from. And obviously the ESP8266 has way less RAM.

daitangio, 2017-09-07:
On my huble opinion, I suggest to update btree documentatiom on ESP8266,  because this error happens very easily. On 1.9.2  I was able to open a database with
` db=btree.open(f, cachesize=2, pagesize=1024 )`
So a small note on how to get it running on ESP8266 is required.

pfalcon, 2017-09-10:
In https://github.com/micropython/micropython/commit/9b4666dad55be2cc978b3dd5da89fd11da0c3cfc , the change is made that the proper errno is raised, the testcase above now raises:

~~~
total: 310 entries (3790 B); db size: 0 B; free heap: 10304 B
total: 320 entries (3916 B); db size: 0 B; free heap: 10320 B
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "bt_ex.py", line 21, in <module>
OSError: [Errno 12] ENOMEM
~~~

pfalcon, 2017-09-10:
Then in 9355cca610e2e7f16a5847271fefe47db97b8cc4, default page size set to 1024 and also minimum cache pages number is decreased. The testcase now runs to completion:

~~~
total: 490 entries (5822 B); db size: 0 B; free heap: 23808 B
total: 500 entries (5938 B); db size: 0 B; free heap: 23824 B
>>> 
~~~

pfalcon, 2017-09-10:
@dpgeorge : I now contemplating switching "cachesize" param to take number of pages instead of bytes, because there's bunch of useless arithmetic in BDB code to convert one to another. We can leave the param name as is, or can rename it to avoid silent issues.

pfalcon, 2017-09-10:
Saves 54 bytes for Xtensa.

dpgeorge, 2017-09-10:
> I now contemplating switching "cachesize" param to take number of pages instead of bytes, because there's bunch of useless arithmetic in BDB code to convert one to another. We can leave the param name as is, or can rename it to avoid silent issues.

As far as I can see, it's just 3 lines it btree/bt_open.c (plus 2 lines to make sure it's not too small).  I'd say it's not worth the effort to patch it just to get rid of this code.  Specifying things in bytes seems more general (eg there could be another DB backend that didn't require cache to be a multiple of page size).  But if you really want to change it then I'd suggest using a different parameter name, like ncache or ncachesize.

pfalcon, 2017-09-16:
@dpgeorge : Thanks for feedback on this.

> As far as I can see, it's just 3 lines it btree/bt_open.c (plus 2 lines to make sure it's not too small).

Yep, something like that, leading to 54 unneeded code bytes on Xtensa.

> I'd say it's not worth the effort to patch it just to get rid of this code. 

Why it caught my attention is because this library goes into "majority" of ports, and there's a fight for every byte (led first of all by me), and 3rd-party libraries then is source of noticeable amount of cruft.

> Specifying things in bytes seems more general (eg there could be another DB backend that didn't require cache to be a multiple of page size).

Yes, bytes are more general. But the idea behind selecting BerkeleyDB was to make a DB-like module which could work with the smallest of ports. To achieve that, it's a "forced" compromise to trade "polymorphism" ("there could be another DB backend") for measures allowing this single backend to work well even in big resource constraints. I.e. trade generality of "interface which can work with any DB" to "DB which works well on any port". Again, that's a compromise, first of all driven by the fact that the code is fully portable, so any optimization or other measure will be readily reusable by all other ports. (That's in contrast to hardware API, where "across-ports" generality is valued higher, because any extra feature added puts additional burden on potentially unlimited number of ports.) So, I'm actually going to document more of btree module (== BerkeleyDB) behavior in docs, and then "bytes vs pages" issue becomes a minor one.

> But if you really want to change it then I'd suggest using a different parameter name, like ncache or ncachesize.

Noted. Also, pondering about it during this time, let's postpone until "code size optimization rush". The latter has the obvious drawback that "if there're breaking changes, it's better to do them sooner than later", but let's save some breakage for later ;-).

pfalcon, 2017-09-16:
> let's postpone until "code size optimization rush"

I've opened https://github.com/micropython/micropython/issues/3319 as a dump of such ideas.

closed 2017-09-16 by pfalcon

pfalcon, 2017-09-17:
> I'm actually going to document more of btree module (== BerkeleyDB) behavior in docs

E.g. 72491b3e40db75e7edf5831e8914a1ca5c8e9937