micropython/micropython #3260: pyboard: SPI slave, how to collect all data till an inter frame gap

franzhollerer, 2017-08-09:
I have a question about the SPI.recv() method, when used as SPI slave.

```c
/// \method recv(recv, *, timeout=5000)
///
/// Receive data on the bus:
///
///   - `recv` can be an integer, which is the number of bytes to receive,
///     or a mutable buffer, which will be filled with received bytes.
///   - `timeout` is the timeout in milliseconds to wait for the receive.
///
/// Return value: if `recv` is an integer then a new buffer of the bytes received,
/// otherwise the same buffer that was passed in to `recv`.
STATIC mp_obj_t pyb_spi_recv(mp_uint_t n_args, const mp_obj_t *pos_args, mp_map_t *kw_args) {
```

Is it possible get all data received till the timeout occured?

Use case: I have a periodic transmission. The length of the frames vary. The inter frame gap is used to determine the end of a frame. I would like to collect all data till the timeout occurs.

Can I use the timeout parameter somehow for this purpose?

dpgeorge, 2017-09-19:
Sorry for the slow response on this issue.

You should be able to access the data that was received before the timeout.  Do it by passing in a buffer, and catching the timeout exception.  The buffer will then contain the received bytes.  There is no way to tell how many bytes were received, but you could prefill the buffer with something like 0xff and then use that to detect how many bytes came in.  Eg:
```python
import errno
buf = bytearray([0xff for _ in range(100)]
try:
    spi.recv(buf, timeout=100)
except OSError as e:
    if e.args[0] == errno.ETIMEDOUT:
        # got a short reply
        pass
    else:
        # some other error, propagate it
        raise
```

Adixy, 2017-09-27:
So I got two pyboards MicroPython v1.9.1-196-g45645043 on 2017-07-29; PYBv1.1 with STM32F405RG communicating with each other via SPI
One pyboard acts as a Slave, the other one as a master

The problem is, the communication works for some short period of time and then something breaks and the comms go nuts. 

Master code:
`py2=pyb.SPI(2, pyb.SPI.MASTER, 500000, polarity=0, phase=1, bits=8, firstbit=pyb.SPI.MSB, ti=False, crc=0x7)

ss=pyb.Pin('Y5', pyb.Pin.OUT_PP)
ss.high()

def py2py():
    ss.low()
    pyb.delay(5)
    data=py2.send_recv(b'\x00\xaa\x00\xbb\x00\xaa\x00\xbb\x00\xaa\x00\xbb\x00\xaa\x00\xbb\x00\xaa\x00\xbb\x00\xaa\x00\xbb')
    ss.high()
    `

Slave code:
`py1 = pyb.SPI(2, pyb.SPI.SLAVE, 500000, polarity=0, phase=1, bits=8, firstbit=pyb.SPI.MSB, ti=False, crc=0x7)
data=bytearray(24)
ss=pyb.Pin('Y5', pyb.Pin.IN)

def f(a):
    py1.send_recv(data)

inter=pyb.ExtInt(ss, pyb.ExtInt.IRQ_FALLING, pyb.Pin.PULL_DOWN, f)
`

And it works for some short period of time, then it just breaks and the Master starts receiving weird buffers. Or, if I also have comms via i2c(1) on Master, the SPI comm completely breaks and Master never receives anything --> data is data=b'\x00'*24

I'm not profesionallist for communications and it's still a new thing for me, so if somebody can help me, please do it :D