micropython/micropython #3261: Q: How to use the pyboard bitmask constants module?

franzhollerer, 2017-08-09:
I would like to use the following constants in a python script:

http://wiki.micropython.org/platforms/boards/pyboard/stm-const-rcc

Which module need I to import? Can please someone provide an example

franzhollerer, 2017-08-09:
supplement. I would like to use this constants to reset the SPI1 peripheral device.

dpgeorge, 2017-08-09:
> Which module need I to import?

You can use the stm module: `import stm` and then something like `stm.mem32[stm.RCC + stm.RCC_AHB1RSTR] |= 0x10` (that's a dummy example).

In general these kinds of questions are best asked at the https://forum.micropython.org

franzhollerer, 2017-08-09:
@dpgeorge I have seen that. My question is how to use the constants defined here:

http://wiki.micropython.org/platforms/boards/pyboard/stm-opmasks-rcc

What I would like to do is something like this:

stm.mem32[stm.RCC + stm.RCC_APB2RSTR] |= stm.RCC_APB2RSTR_SPI1RST

But I failed to fiugre out how to access RCC_APB2RSTR_SPI1RST.

franzhollerer, 2017-08-09:
> In general these kinds of questions are best asked at the https://forum.micropython.org

Sorry, I have overlooked this forum. Will use it next time.

dpgeorge, 2017-08-11:
> But I failed to fiugre out how to access RCC_APB2RSTR_SPI1RST.

I see.  Those bit-level constants are not provided in the MicroPython build, mainly because there are so many of them.  If you need them then please either define them in your script, or just use the raw values in the code.

closed 2017-08-11 by dpgeorge