micropython/micropython #3263: upip_bootstrap script is broken

boneskull, 2017-08-12:
`wget` (on macOS, installed via Homebrew) doesn't want to grab the tarball off of pypi.

I don't quite understand the necessity of downloading the tarball when the upip scripts are actually in the `tools/` dir... perhaps it should be removed?

pfalcon, 2017-08-23:
Update the title to correspond to the bug report content.

pfalcon, 2017-08-23:
@dpgeorge : What's your stance on tools/upip_bootstrap.sh script? On one hand, we now ship upip builtin, so could remove that bootstrap script. On the other hand, builtin upip is just one possible configuration (even if one that was made default). So, there's nothing wrong with building upy without frozen upip, and then bootstrapping it separately (and the script serves as a good executable documentation).

boneskull, 2017-08-23:
my 2c: pick one way and go with it, because it's kind of confusing otherwise

pfalcon, 2017-08-23:
The "one way" is described in the README, so there's little chance to be confused. And developer tools are developers tools, and serve their own purpose.

boneskull, 2017-08-23:
if there was little chance to be confused I wouldn't have posted the issue 😐

dpgeorge, 2017-09-06:
> What's your stance on tools/upip_bootstrap.sh script?

As @boneskull points out, the upip code is available in tools/ so a much simpler bootstrapping would be to just copy out those files to the relevant installation directory.  And such a bootstrapping is even relevant to bare-metal, by installing to the device.

So I'd have slight preference to remove bootstrap_upip.sh.

dpgeorge, 2022-06-23:
`bootstrap_upip.sh` was removed in c2b5bfcc0c6648f549af215f3d2f47302e962c96

closed 2022-06-23 by dpgeorge