micropython/micropython #3265: build fails on unix without gnu find

rofl0r, 2017-08-12:
```
micropython-1.9.1/unix $ make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
find: unrecognized: -L
BusyBox v1.20.2 multi-call binary.

Usage: find [PATH]... [OPTIONS] [ACTIONS]

...

make: *** No rule to make target `lib/berkeley-db-1.xx/btree/bt_close.c', needed by `build/genhdr/qstr.i.last'.  Stop.

```

jimmo, 2017-08-13:
Just to add a bit more information -- can you please confirm that changing the corresponding line of py/mkrules.mk to use `-follow` (which is deprecated, but should be supported by both busybox and GNU find) works for you.

`FROZEN_MPY_PY_FILES := $(shell find $(FROZEN_MPY_DIR) -follow -type f -name '*.py' | $(SED) -e 's=^$(FROZEN_MPY_DIR)/==')`

rofl0r, 2017-08-13:
yes, that syntax works (in that find doesn't bomb out).
i still do get the other error 
```
No rule to make target `lib/berkeley-db-1.xx/btree/bt_close.c'
``` 
though, and when i start with making axtls as described in the README, i get:
```
You cloned without --recursive, fetching submodules for you.
(cd ..; git submodule update --init --recursive)
fatal: Not a git repository (or any of the parent directories): .git
make: *** [../lib/axtls/README] Error 128

```

this is using the last pseudo-release tarball using the github tag v1.9.1.
that means if you want to supply some means for people to download *complete* release tarballs, you shouldn't use the "automatic tarball download from tag" feature of github, [but attach a real release tarball](https://github.com/sabotage-linux/sabotage/wiki/Why-github-downloads-suck).

btw, i've seen here in an issue that axtls has incomplete support for some TLS features, is it possible to build mbedtls instead ?

dpgeorge, 2017-08-15:
I don't know if there is a way to fix this issue to be compatible with all versions of find.  In the end maybe it's best to remove symlinks all together and provide a different way to specify .py sources to be included as frozen bytecode (see also https://github.com/micropython/micropython-esp32/issues/142#issuecomment-322023040)

> that means if you want to supply some means for people to download complete release tarballs, you shouldn't use the "automatic tarball download from tag" feature of github

This issue was raised previously in #2952.  I don't think it's possible to disable the automatic tarball download and attaching them manually is extra effort.  The best solution is to just use git to get the source code.

> i've seen here in an issue that axtls has incomplete support for some TLS features, is it possible to build mbedtls instead ?

Yes you can build unix with mbedtls, if you have mbedtls installed separately (you'll need to work that bit out for yourself).

dpgeorge, 2020-06-09:
Frozen code is now specified by a manifest and no longer relies on symlinks.  So this issue should be fixed.

dpgeorge, 2020-06-09:
See e81f538e2509938b25eb48ed2670142eed1f2573

closed 2020-06-09 by dpgeorge