micropython/micropython #3266: ubinascii: a2b_base64(...) behavior is inconsistent with CPython

ccooper21, 2017-08-12:
The `a2b_base64(...)` function in the `ubinascii` module raises an exception if its byte string argument contains any unexpected characters.  In comparison to CPython, the CPython implementation ignores all bytes representing new line, carriage return, and space characters, as well as all bytes whose most significant bit is set (i.e. bytes representing extended 8-bit ASCII characters).  This behavior can be seen in the CPython implementation [here](https://github.com/python/cpython/blob/master/Modules/binascii.c#L468:470).

Aside from the current behavior being inconsistent with CPython, this creates an oddity in the MicroPython `ubinascii` module in that the `b2a_base64(...)` and `a2b_base64(...)` inverse functions are incompatible.  Specifically, the `b2a_base64(...)` function returns a Base64 representation with a trailing new line character.  While this is consistent with this function's CPython behavior, passing this result to the `a2b_base64(...)` function fails because the trailing new line character is unexpected.  Logically, it should be possible to cyclically pass the result of either function into its inverse function without failure.

ccooper21, 2017-08-12:
I'd be willing to take a stab at fixing this, but I have a few questions:

- Is this actually a defect, or is the current behavior by design?  Asked another way, would this change be considered for merging?

- What is the best approach to handle this?  One approach is to preprocess the input byte string to strip the bytes representing characters to be ignored.  This would be more data ineffecient at run-time, but require minimal change to the existing code and be straightforward.  Alternatively, the existing implementation could be altered to ignore these bytes, but on first glance that appears to involve some non-trivial rework.  I'm not sure which approach would have the least impact to the resulting byte code size.

robert-hh, 2017-08-13:
Please have a look at that pull request: https://github.com/micropython/micropython/pull/3254
It may already cover your issue.

ccooper21, 2017-08-13:
Thanks for pointing out this pull request!  It does indeed appear to address the issue that I identified.  Hopefully those involved can work through the remaining concerns, and get it merged soon.

pfalcon, 2017-08-13:
> Is this actually a defect, or is the current behavior by design? Asked another way, would this change be considered for merging?

Those are good questions to pose. The default answer for 1st question is usually: "By design, unless proven to a bug." And for 2nd - "Yes, if there're better reasons to change it, than those to keep it as is". Mind that (full) CPython compatibility compatibility in general, far less "bug to bug" compatibility is non-goal for MicroPython: https://github.com/micropython/micropython/wiki/ContributorGuidelines .

For the discussion of specific feature, let's move to #3254.

closed 2017-08-13 by pfalcon