micropython/micropython #3268: unix port: blocking IO bug exposed

pmp-p, 2017-08-14:
- happens more often on arm 32bits
- micropython is  quite unreliable on long IO run loops. ( max ~ 100MB traffic 32k packet size on arm).
- python3 doesn't bug on that.

```

"""
RANDOM OUTPUT:

 ------- ok ----------
return
 ------- maybe ok  ----------
return
 ------- or not ----------
return
Unhandled exception in thread started by <bound_method e716120 <subprocess object at e716100>.<function run at 0xe7160a0>>
Traceback (most recent call last):
  File "abug.py", line 14, in run
OSError: 4

"""

import sys
sys.path.append('.')

import _thread
import time
import gc
import os

class subprocess:

    def __init__(self):
        self.thr = _thread.start_new_thread( self.run , () )

    def run(self):
        sys.stdin.read(1)

    def read(self):
        global bugme
        if bugme:
            gc.collect() #  <<<< help force the bug show up now , OSError: 4 , but still would still happen cyclicly
        return 'return'

bugme = False
sp=subprocess()

print(" ------- ok ---------- ")
print(  sp.read() )

bugme= True

print(" ------- maybe ok  ---------- ")
print(  sp.read() )

time.sleep(1)

print(" ------- or not ---------- ")
print(  sp.read() )

time.sleep(2) #let time to display error

```

pfalcon, 2017-08-14:
Let me update the title to what I see in your report.

dpgeorge, 2017-08-15:
The reason that gc.collect() forces the bug to show up is because, with threading enabled, it sends SIGUSR1 to the independent threads to get them to scan their local registers and stack.  And that signal interrupts the read() call.

dhylands, 2017-08-15:
This page: http://man7.org/linux/man-pages/man7/signal.7.html
under the section: "Interruption of system calls and library functions by signal handlers" and the section after that, lists all of the system calls which can return EINTR. Several of the linux C runtime libraries (like glibc) provide a macro named: TEMP_FAILURE_RETRY which will call the system call in a loop until something other than EINTR is returned.

pmp-p, 2018-01-23:
maybe related material there https://www.python.org/dev/peps/pep-0475/

dpgeorge, 2020-03-27:
PEP 475 (retry in case of EINTR) was implemented in 9418611c8aa860812e28b6e2b6bfde7be5817b43.  But I'm not sure that would fix the case here of `gc.collect()` in another thread interrupting a syscall.

dlech, 2020-03-27:
This case is exactly the motivation for this change, so yes, it should be fixed.

dpgeorge, 2020-06-09:
> so yes, it should be fixed.

Ok, let's close this issue then.

closed 2020-06-09 by dpgeorge