micropython/micropython #3269: RFC: OSError: Make sure it's always called with (numeric) errno

pfalcon, 2017-08-14:
This was a direction we were already moving for some time, seemingly under silent agreement. To avoid "d'oh" moments when some parties figure out there was no agreement, this explicit ticket is submitted.

Proposal: OSError should always take an errno argument, usages non conforming should be converted (either to use another exception, or to use errno).

Foundations:

From CPython docs:

https://docs.python.org/3/library/exceptions.html#BaseException.args

> Some built-in exceptions (like OSError) expect a certain number of arguments and assign a special meaning to the elements of this tuple [`args`]

https://docs.python.org/3/library/exceptions.html#OSError

> exception OSError([arg])
> exception OSError(errno, strerror[, filename[, winerror[, filename2]]])

That's CPython, we being Micro and Clean, can reduce that to the strict requirement of `exception OSError(errno)`.

Alternatives:

An alternative might be to allow form of `OSError(errno, [strerror])`, i.e. where string msg is optional 2nd argument, and overrides default message for errno, but IMHO, that's too bloated and we should try hard to fir within just `OSError(errno)` (or use other exceptions).

dpgeorge, 2017-08-15:
> Proposal: OSError should always take an errno argument,

In general I think this is a good idea (errors are something that shouldn't usually happen and error messages only given to aid the user), but in practice I think there are some cases which won't fit this rule.  The cases to convert are (only the message string is given here):
```
can't create thread
pin does not have IRQ capabilities
UART(1) can't read
scan failed
I2C operation not supported
stream operation not supported
accelerometer not found
no available NIC
failed to init CC3000 module
could not connect to ssid=%s, sec=%d, key=%s
unable to load libjvm.so, use LD_LIBRARY_PATH
unable to create JVM
[addrinfo error %d]
```
And there's a few in .py code as well, eg:
```
nRF24L01+ Hardware not responding
can't set 512 block size
EOF in headers
Not a websocket request
```

> An alternative might be to allow form of OSError(errno, [strerror])

I'd say this is the way to go: require the first arg to be an appropriate errno number and then an additional arg can be a specific string if warranted (and only for the cases that really help the user).

> or use other exceptions

I don't think there's anything else suitable.  RuntimeError is the closest but that should be reserved for internal VM/runtime errors (same with SystemError).

pfalcon, 2017-08-15:
My idea was that quite a few can be converted to TypeError, ValueError, etc. But indeed that has limits, e.g. I had in mind to convert to TypeError:

~~~
        || ((flags & MP_STREAM_OP_READ) && stream_p->read == NULL)
        || ((flags & MP_STREAM_OP_WRITE) && stream_p->write == NULL)
        || ((flags & MP_STREAM_OP_IOCTL) && stream_p->ioctl == NULL)) {
        // CPython: io.UnsupportedOperation, OSError subclass
        mp_raise_msg(&mp_type_OSError, "stream operation not supported");
~~~

But as comment says, it must be an OSError.

> I'd say this is the way to go: require the first arg to be an appropriate errno number and then an additional arg can be a specific string if warranted (and only for the cases that really help the user).

Sounds good then (need to think whether to extend mp_raise_OSError() or add a separate func).

> I don't think there's anything else suitable. RuntimeError is the closest but that should be reserved for internal VM/runtime errors (same with SystemError).

Note that https://docs.python.org/3/library/exceptions.html#RuntimeError says:

> Raised when an error is detected that doesn’t fall in any of the other categories.

Though we all somehow got used that it's interpreter's runtime error. Maybe we should nudge that stereotype a bit ;-).

pfalcon, 2017-08-15:
@candeira : FYI

candeira, 2017-08-15:
Thanks, guys. I'll get on it over the weekend.

dpgeorge, 2020-06-09:
Most `OSError`s now use just a numeric argument.  And in cases where more info should be passed to the user it's of the form `OSError(errno, str)`, see eg 1ae7e0e561134c39471e3000eced33da732b4103

So I'll close this issue.

closed 2020-06-09 by dpgeorge