micropython/micropython #3272: PinAF seems to be missing a method.

mwm, 2017-08-15:
The documentation describes __str__(), but it's not on the objects I get from Pin.af_list in b6a32.

To further confuse matters, the first two PinAF objects I get back are identical as far as I can tell:

```
>>> a1, a2, *_ = pyb.Pin.cpu.A0.af_list()
>>> a1.name() == a2.name()
True
>>> a1.reg() == a2.reg()
True
>>> a1.index() == a2.index()
True
>>> dir(a1)
['index', 'name', 'reg']
>>> a1 == a2
False
```

But as you can see, the interpreter found some difference between them.

Looking at the _af.csv file, I can see that those two are TIM2_CH1 and TIM2_ETR. The difference is apparently CH1 vs. ETR, which I'd actually like to be able to see.

dhylands, 2017-08-15:
This really belongs on the forum. The same AF mode does both CH1 and ETR. I wanted them to be separated so that you didn't have to have a different constant for CH1 on some timers and CH1 on others.

ETR mode is an input feature, and whether its active or not is really controlled by registers.

mwm, 2017-08-15:
Ok. Opened at https://forum.micropython.org/viewtopic.php?f=2&t=3687

pfalcon, 2017-10-28:
If the method is missing, please update the title with the relevant info.

closed 2022-06-23 by dpgeorge