micropython/micropython #3273: mp_sched_schedule's scheduled callbacks don't run on empty REPL lines

pfalcon, 2017-08-15:
Ok, so using mp_sched_schedule() with interactive REPL is problematic - scheduled stuff won't run until REPL executes something. That's understood. However, it would be nice if empty lines entered actually caused them to be executed. Otherwise, it requires really dirty workaround of typing random expressions to get it going, like me typing `1` below:

~~~
>>> conn.gatt_read(3, print)
>>> read_cb: 0
4d 49 

>>> 1
b'MI'
1
>>> 1
1
>>> conn.gatt_read(5, print)
>>> read_cb: 0
00 00 

>>> 1
b'\x00\x00'
1
>>> conn.gatt_read(7, print)
>>> read_cb: 0
00 

>>> 
>>> 1
b'\x00'
1
>>> conn.gatt_read(9, print)
>>> read_cb: 0
06 00 50 00 00 00 f4 01 

>>> 1
b'\x06\x00P\x00\x00\x00\xf4\x01'
1
~~~

dpgeorge, 2017-08-16:
What platform/port do you use?  It's generally up to the port to make this behave correctly and IIRC it works OK on esp8266 and pyboard.  If you could give minimal code that reproduces the error that would help.

pfalcon, 2017-08-16:
It's Zephyr port, and my first attempt to use mp_sched_schedule() at all. There's no immediate way to reproduce the above (it's prototype of BLE module). I'll look into what stmhal/esp8266 do, but from the above I can imagine that probably lexer detects that input string is empty and doesn't call VM. It could at least signal it to process scheduled callbacks in this case, and that would be port-independent.

dpgeorge, 2017-08-16:
> I'll look into what stmhal/esp8266 do,

They call mp_handle_pending() in their MICROPY_EVENT_POLL_HOOK macro, which is in turn called from the busy-wait loop when waiting for incoming chars for the REPL.  With this approach, scheduled callbacks will be called while waiting for a uselect.poll.poll() call, for example.

pfalcon, 2017-08-16:
Thanks, will look into adding that.

dpgeorge, 2021-04-14:
This behaviour of `mp_sched_schedule()` is intended, so closing.

dpgeorge, 2021-04-28:
Ah, this didn't get closed... anyway, with #7177 scheduled events will be run asynchronously at the REPL and during sleeps.

dpgeorge, 2021-04-30:
Now works via d120859857e9e34bded146977f18a2be8c53ce7b

closed 2021-04-30 by dpgeorge