micropython/micropython #3274: PYB: soft reboot -> Machine: soft reboot?

mcauser, 2017-08-16:
I just noticed all of the ports print `PYB: soft reboot` when you perform a soft `Ctrl-D` reboot. eg.

```
PYB: soft reboot
#7 ets_task(40100164, 3, 3fff837c, 4)
OSError: [Errno 2] ENOENT
MicroPython v1.9.1-224-gace9fb54-dirty on 2017-08-16; ESP module with ESP8266
Type "help()" for more information.
```

Should that be updated to `Machine: soft reboot` at some point?

dpgeorge, 2017-08-16:
"PYB: soft reboot" is a special string that tools/pyboard.py searches for to sync with the board.  But actually it doesn't care about the "PYB:" part so that can be changed to whatever a port likes.  On the pyboard PYB is used because that's the name of the board (it doesn't have anything to do with the pyb module).  If it were to be changed, something like "UPY" or "MPY" would be better.

aykevl, 2017-10-19:
This will soon be added to the nrf port - which prefix should we use there? I'm leaning towards "MPY: soft reboot".

dpgeorge, 2017-10-20:
> I'm leaning towards "MPY: soft reboot".

Yes that's OK.

dpgeorge, 2019-02-12:
PYB was changed to MPY in 812969d615e130bef7e525553dbb7b01c27191ea

closed 2019-02-12 by dpgeorge