micropython/micropython #3278: cut & paste into repl swallows '-'

fluffymcpuffin, 2017-08-18:
MicroPython v1.9.1-399-g40ae6724 running on an ESP32
when i cut 10/5 - 20/10 from my editor and paste it into repl all is good.
but when i cut (x/5) – (y/10) and paste it into repl the minus sign has dissapeared.
all works well under python3 in terminal mode.

MicroPython v1.9.1-399-g40ae6724 on 2017-08-15; ESP32 module with ESP32
>>> x,y = 10,20
>>> 10/5 - 20/10
0.0
>>> (x/5)  (y/10)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: 'float' object is not callable
>>>

pfalcon, 2017-08-18:
> but when i cut (x/5) – (y/10) 

This isn't a minus sign, but some other, non-ascii character.

robert-hh, 2017-08-19:
And Python3 in terminal mode complains too:
```
  File "<stdin>", line 1
    (x/5) – (y/10)
          ^
SyntaxError: invalid syntax
```

fluffymcpuffin, 2017-08-19:
hi robert,

you are absolutely right, thanks for you response....
you guys are great,
so long
juergen

closed 2017-08-19 by fluffymcpuffin