micropython/micropython #3279: TLS buffer overflow

cemox, 2017-08-19:
Hi, I wrote a program that reads content from a news api request, everything works fine (actually I orijinally built my esp8266 projest with nodemcu but it is useless whit hppts connection, micropython is great).

in my program I have ssl.wrap_socket (called s) 
`s = ssl.wrap_socket(s, server_hostname=TS_HOST)`
and it reads 2048 bytes from stream 
`data = s.read(2048)`

it works most of the time, but at times, it gives me buffer overflow error like this 

> TLS buffer overflow, record size: 5296 (+5)

 when the incoming data is huge. Isn't the size parameter of read method supposed to force the wrapper to stop reading data when it reaches 2048 bytes and ignore the rest and continue execution without throwing an exception?

dpgeorge, 2017-08-21:
> Isn't the size parameter of read method supposed to force the wrapper to stop reading data when it reaches 2048 bytes and ignore the rest and continue execution without throwing an exception?

With TLS (https) it's a bit more complicated than that: the underlying TLS communication stream has a maximum packet size of 16k and it's possible that your device is receiving such a packet.  This packet cannot be split up into chunks, it must be recieved, stored and processed (by the axtls library) all at once.  The esp8266 must therefore have enough buffer space to store 16k at once.  This is independent from your request for 2048 bytes from the stream, which is a request for unencrypted data.

You can try to work around the issue by making sure the incoming data is not that large.

Otherwise it's possible to recompile the esp8266 firmware to allow for a larger buffer size.  Edit line 231 of esp8266/Makefile and change the `-DRT_EXTRA=4096` to something like `-RT_EXTRA=5120`, then do `make clean` and `make`.

cemox, 2017-08-21:
Very informative answer, thank you.

closed 2017-08-21 by cemox

cefn, 2018-01-30:
Is there support in Axtls/Micropython for RFC6606 - negotiating lesser max_fragment_size values for a TLS connection? 

With this approach the esp8266 build would attempt to negotiate with the server a maximum TLS record size corresponding to its buffer, (though of course this only works if the esp8266 is a client and the server could always refuse). 

Do you know if this negotiation already takes place, or if perhaps it could be added? I struggled to find any part of the micropython build which attempts to negotiate TLS record sizes. That is, assuming the server support for RFC6606 is widespread enough to bother.

balajisraghavan, 2019-02-09:
> Otherwise it's possible to recompile the esp8266 firmware to allow for a larger buffer size. Edit line 231 of esp8266/Makefile and change the `-DRT_EXTRA=4096` to something like `-RT_EXTRA=5120`

Could you please point out the place to increase the TLS buffer size in the latest codebase? Can't find `-DRT_EXTRA=4096` in the make file anymore

cefn, 2019-02-10:
Looks to me like it's at https://github.com/micropython/micropython/blob/master/ports/esp8266/Makefile#L8 but maybe I missed something.

dpgeorge, 2019-02-13:
Yes, line 8 of the esp8266/Makefile is the new location.

balajisraghavan, 2019-02-17:
Thanks @cefn and @dpgeorge. I tried changing it but looks like google cloud certs are about 8k and the esp8266 cant cope with that. Planning to try with the esp32 now.

VinzSpring, 2019-07-09:
> With TLS (https) it's a bit more complicated than that: the underlying TLS communication stream has a maximum packet size of 16k and it's possible that your device is receiving such a packet. This [...]

Hi, 
I am trying to build the ROM with the specified buffer size for use with the telegram bot API (4MB ESP8266). However I can't get it to compile successfully since the esp-open-sdk fails to build. 
Would someone mind to provide a link to a valid binary with increased buffersize?
I really need to be able to use the telegram API.
Thank you so much.

dpgeorge, 2019-07-09:
> However I can't get it to compile successfully since the esp-open-sdk fails to build.

You can download a compiled version of the esp-open-sdk from https://github.com/jepler/esp-open-sdk/releases/download/2018-06-10/xtensa-lx106-elf-standalone.tar.gz

YoruTen, 2020-07-08:
Have anyone managed to create a build with a larger buffer? I tried to do a 8192 buffer but got "fatal exception 29(storeprohibited cause)" as a new error. I would guess that 8192 is way to large for the esp8266, but my TLS buffer overflow error reported a size of 7500.

cefn, 2020-07-08:
> A pre-configured image (based on version 5d0d12c9) which works around the ESP8266 limitations noted below is available at http://shrimping.it/project/medea/ . It is based on the latest github at the time of writing with the medea library and examples onboard as frozen modules and a 8192byte TLS buffer to allow larger HTTPS payloads (e.g. at least 10 tweets from a Twitter timeline, at least a day's worth of 3-hourly OpenWeatherMap forecasts).
https://github.com/ShrimpingIt/medea

Possibly so old as to be useless though.

flokain, 2020-10-21:
@YoruTen i was able to set the buffersize to 8192,b as the packages i recieved were 7515b. It worked without a problem.

YoruTen, 2021-01-13:
Which board are you using and could you post the image?

LLllIIUU, 2022-03-11:
> Which board are you using and could you post the image?
> […](#)
> -- //Andreas

Hi, have you successfully set the buffer size to 8192b now? What if it was successful?

flokain, 2022-03-11:
@YoruTen Sorry for the late reply. I was using a wemos d2 mini board. the image consist a custom firmware so it would not work for you. but find the script to build it at the end of the post.

I used the following make command to create the image:

```bash
 make AXTLS_DEFS_EXTRA="-Dabort=abort_ -DRT_MAX_PLAIN_LENGTH=1024 -DRT_EXTRA=8192"
```

bonus: in case you need more memory i recommend sth like the following tweak BEFORE you compile the image:

```bash
sed -i 's/STATIC char heap.*/STATIC char heap[44 * 1024];/g'  micropython/ports/esp8266/main.c
```
bonus: in case you need more firmware storage i recommend sth like the following tweak BEFORE you compile the image:
```bash
sed -i 's/    irom0_0_seg :  org = 0x40209000, len = .*/    irom0_0_seg :  org = 0x40209000, len = 0xa7000/g' micropython/ports/esp8266/boards/esp8266.ld
```

if you are interested how i solved to automate those tedious tasks for building a custom firmware for my own project[ take a look at this script
](https://github.com/flokain/Smart-Cube/blob/micropython/build_deploy.sh)

hope that helps