micropython/micropython #3280: Unix (only) socket.getaddrinfo() fails

peterhinch, 2017-08-20:
This would appear to be a regression. Built ffrom today's source.
Unix:
```
$ upython
MicroPython v1.8.7-1080-g394c536-dirty on 2017-08-20; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import usocket as socket
>>> socket.getaddrinfo('iot.eclipse.org', 1883)[0][-1]
bytearray(b'\x02\x00\x07[\xc6)\x1e\xf1\x00\x00\x00\x00\x00\x00\x00\x00')
>>> 
```
ESP8266:
```
MicroPython v1.8.7-1080-g394c536-dirty on 2017-08-20; ESP module with ESP8266
Type "help()" for more information.
>>> 
>>> ipm open,type:2 0

>>> import usocket as socket
>>> socket.getaddrinfo('iot.eclipse.org', 1883)[0][-1]
('198.41.30.241', 1883)
>>> 
```

pfalcon, 2017-08-20:
So, where's regression?

peterhinch, 2017-08-20:
A search of previous issues suggests it has worked recently on Uix, e.g. @dpgeorge's test report on https://github.com/micropython/micropython/issues/3164.

pfalcon, 2017-08-23:
@peterhinch : Unless you can elaborate what you mean by "getaddrinfo fails", I would close this as invalid. Thanks for understanding. Additional info: https://www.chiark.greenend.org.uk/~sgtatham/bugs.html

peterhinch, 2017-08-23:
Sorry if I'm being dumb but under CPython, ESP8266 MicroPython and ESP32 MicroPython I get the expected result ('198.41.30.241', 1883).

Under Unix MicroPython I get a bytearray. Am I missing something obvious here?

pfalcon, 2017-08-23:
That's how it has been all the time. Unix port is a bastion of effectiveness in MicroPython, and if you think, it doesn't make much sense to take a binary address structure from C's getaddrinfo(), convert it to Python tuple with string and integer, to convert them back to binary structure on the next step, on a call to any other socket API function, like connect() etc. - having wasted few memory and bunch of cycles in this useless process. So, Unix port doesn't. Again, that's how it was all the time, and now described at http://docs.micropython.org/en/latest/pyboard/library/usocket.html#socket-address-format-s

closed 2017-08-23 by pfalcon

peterhinch, 2017-08-23:
Thanks for clarifying - apologies for raising a non-issue.