micropython/micropython #3281: Repeated keyword argument should cause SyntaxError, not TypeError

alex-robbins, 2017-08-22:
CPython:
```python
>>> def f(x): pass
... 
>>> f(x=1, x=1)
  File "<stdin>", line 1
SyntaxError: keyword argument repeated
```
MicroPython:
```python
>>> def f(x): pass
... 
>>> f(x=1, x=1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: function got multiple values for argument 'x'
```

Nitpicky? Yes. But I doubt fixing this would change code size, so...

See also #466. I would have submitted a patch instead of an issue, but it looks like that exception can be raised from two different places, and I don't know the difference.

Thanks,
Alex

dpgeorge, 2017-08-23:
Thanks for the report.  There are actually 2 different ways to get this error: the SyntaxError way from `f(x=1, x=1)`, and the TypeError way from `f(x=1, **{'x':1})`.

The latter can only be caught at runtime, hence it's a TypeError, and that's the one uPy checks for.

The former case can be caught at compile time and be a SyntaxError but uPy currently doesn't perform such checking.  It would cost a bit of extra code, and slow the compiler down, to make N^2 passes over function calls to check that there are no duplicate keyword args.

pfalcon, 2017-08-23:
> It would cost a bit of extra code, and slow the compiler down

That's what I thought too. We'd better leave it as is.

@alex-robbins, if you care, you can add this case to http://docs.micropython.org/en/latest/pyboard/genrst/index.html

dpgeorge, 2017-08-24:
A strong argument for fixing this is that it's always better to find errors at compile time rather than wait until run time.  But whether it's worth the cost in this case is hard to say... probably it's not.

pfalcon, 2017-08-24:
We can enable it for mpy-cross then, one sweet day... ;-)

alex-robbins, 2017-08-25:
I only submitted this because I figured it would be really easy to fix with no downsides to speak of. Didn't realize that the TypeError was correct in some cases. Feel free to close this, as far as I'm concerned.