micropython/micropython #3283: stmhal: Pull ST HAL and ST CMSIS files out into a separate repo, as a git submodule

dpgeorge, 2017-08-23:
As discussed in #1907 comment https://github.com/micropython/micropython/issues/1907#issuecomment-322367094 I plan to move all the files in `stmhal/hal` and `stmhal/cmsis` to a separate repository and then include them back in this repository as a git submodule.

It's pretty clear what to do and straightforward to do it, namely using a "vendor branch" workflow (see eg discussion here: https://github.com/micropython/micropython/pull/2140#issuecomment-242359550 and this repo for an example: https://github.com/micropython/oofatfs) to easily upgrade the HAL and apply our custom patches to it.

The main question is what to name the repo and where to link it into this one.  I suggest something like `stmlib` and so the repo would be `micropython/stmlib.git` and would be linked as a git submodule at `lib/stmlib/` in this repo.

pfalcon, 2017-08-23:
Sounds good to me.

dpgeorge, 2017-09-04:
This was done, see PR linked above.

closed 2017-09-04 by dpgeorge