micropython/micropython #3285: Compiling on termux (clang as gcc) fails

ASMfreaK, 2017-08-28:
I've git cloned the repo, run `make axtls` and then `make` to see this error:

```
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build/genhdr
Generating build/genhdr/mpversion.h
GEN build/genhdr/qstr.i.last
GEN build/genhdr/qstr.split
GEN build/genhdr/qstrdefs.collected.h
QSTR updated
GEN build/genhdr/qstrdefs.generated.h
CC ../py/mpstate.c
CC ../py/nlrx86.c
../py/nlrx86.c:40:5: error: macro expansion producing 'defined'
      has undefined behavior [-Werror,-Wexpansion-to-defined]
#if NLR_OS_WINDOWS
    ^
../py/nlrx86.c:38:25: note: expanded from macro
      'NLR_OS_WINDOWS'
#define NLR_OS_WINDOWS (defined(_WIN32) || defined(__CYGWIN__))
                        ^
../py/nlrx86.c:40:5: error: macro expansion producing 'defined'
      has undefined behavior [-Werror,-Wexpansion-to-defined]
../py/nlrx86.c:38:44: note: expanded from macro
      'NLR_OS_WINDOWS'
#define NLR_OS_WINDOWS (defined(_WIN32) || defined(__CYGWIN__))
                                           ^
2 errors generated.
make: *** [../py/mkrules.mk:47: build/py/nlrx86.o] Error 1
```
I have this version of compiler:
```
$ gcc -v
clang version 4.0.0 (tags/RELEASE_400/final)
Target: i686--linux-android
Thread model: posix
InstalledDir: /data/data/com.termux/files/usr/bin
```

stinos, 2017-08-28:
nlrx86.c should be changed so that it has
```
#if(defined(_WIN32) || defined(__CYGWIN__)
#define NLR_OS_WINDOWS 1
#else
#define NLR_OS_WINDOWS 0
#endif
```

dpgeorge, 2017-08-29:
Thanks for the report, fixed in be8e5744e64aec10a3499fd6ea034bbf4be0c871

closed 2017-08-29 by dpgeorge