micropython/micropython #3286: More compilation errors on termux (x86 android, clang)

ASMfreaK, 2017-08-29:
```
python ../py/makeversionhdr.py build/genhdr/mpversion.h
CC gccollect.c
gcc -I. -I.. -Ibuild -Wall -Werror -Wpointer-arith -Wuninitialized -std=gnu99 -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -Os   -fdata-sections -ffunction-sections -fno-asynchronous-unwind-tables -c -MD -o build/gccollect.o gccollect.c
gccollect.c:83:14: error: variable 'ebx' is uninitialized when
      used here [-Werror,-Wuninitialized]
    arr[0] = ebx;
             ^~~
gccollect.c:79:22: note: initialize the variable 'ebx' to
      silence this warning
    register long ebx asm ("ebx");
                     ^
                      = 0
gccollect.c:84:14: error: variable 'esi' is uninitialized when
      used here [-Werror,-Wuninitialized]
    arr[1] = esi;
             ^~~
gccollect.c:80:22: note: initialize the variable 'esi' to
      silence this warning
    register long esi asm ("esi");
                     ^
                      = 0
gccollect.c:85:14: error: variable 'edi' is uninitialized when
      used here [-Werror,-Wuninitialized]
    arr[2] = edi;
             ^~~
gccollect.c:81:22: note: initialize the variable 'edi' to
      silence this warning
    register long edi asm ("edi");
                     ^
                      = 0
gccollect.c:86:14: error: variable 'ebp' is uninitialized when
      used here [-Werror,-Wuninitialized]
    arr[3] = ebp;
             ^~~
gccollect.c:82:22: note: initialize the variable 'ebp' to
      silence this warning
    register long ebp asm ("ebp");
                     ^
                      = 0
4 errors generated.
make[1]: *** [../py/mkrules.mk:47: build/gccollect.o] Error 1
make[1]: Leaving directory '/data/data/com.termux/files/home/projects/micropython/mpy-cross'
make: *** [../py/mkrules.mk:107: ../mpy-cross/mpy-cross] Error 2
```
This is what make generated.

dpgeorge, 2017-08-30:
Try removing "-Werror" from the makefile and see how much further you can get.

stinos, 2017-08-30:
@ASMfreaK can you state what version of clang you use?
Also, does this mean clang by default enables more warnings than gcc? But only this version of clang and not the one used on OSX?

ASMfreaK, 2017-08-30:
@stinos
```bash
$ gcc --version       
clang version 4.0.0 (tags/RELEASE_400/final)                   Target: i686--linux-android                                    Thread model: posix
InstalledDir: /data/data/com.termux/files/usr/bin
```
@dpgeorge
I'll try.

ASMfreaK, 2017-08-30:
@dpgeorge I've disabled -Werror in unix and mpy-cross Makefiles:
```
make[1]: Leaving directory '/data/data/com.termux/files/home/projects/micropython/mpy-cross'
MPY modules/upip.py                                            mkdir -p build/frozen_mpy/                                     ../mpy-cross/mpy-cross -o build/frozen_mpy/upip.mpy -s upip.py -mcache-lookup-bc modules/upip.py                              make: *** [../py/mkrules.mk:117: build/frozen_mpy/upip.mpy] Segmentation fault
```
I've edited `../py/mkenv.mk` with `DEBUG :=` and recompiled with debug:
```
make clean  && make axtls DEBUG=1 && make DEBUG=1 V=1
```
```
gdb --args ../mpy-cross/mpy-cross -o build/frozen_mpy/upip.mpy -s upip.py -mcache-lookup-bc modules/upip.py                                           

GNU gdb (GDB) 8.0                                              
Copyright (C) 2017 Free Software Foundation, Inc.              
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>                                                 
This is free software: you are free to change and redistribute it.                                                            
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"                                                    and "show warranty" for details.                               
This GDB was configured as "i686-linux-android".               
Type "show configuration" for configuration details.           For bug reporting instructions, please see:                    <http://www.gnu.org/software/gdb/bugs/>.                       
Find the GDB manual and other documentation resources online at:                                                              <http://www.gnu.org/software/gdb/documentation/>.              For help, type "help".                                         Type "apropos word" to search for commands related to "word"...

Reading symbols from ../mpy-cross/mpy-cross...(no debugging symbols found)...done.                                            
(gdb) run                                                      
Starting program: /data/data/com.termux/files/home/projects/micropython/mpy-cross/mpy-cross -o build/frozen_mpy/upip.mpy -s upip.py -mcache-lookup-bc modules/upip.py                                                                                      
Program received signal SIGSEGV, Segmentation fault.           0x56558747 in ?? ()

(gdb) bt full                                                  
#0  0x56558747 in ?? ()                                       
No symbol table info available.                                
#1  0x3424448b in ?? ()                                        No symbol table info available.                                #2  0x04244489 in ?? ()                                        No symbol table info available.                                #3  0xfe0c838d in ?? ()                                        No symbol table info available.                                #4  0x0489ffff in ?? ()                                        No symbol table info available.                                #5  0x4d63e824 in ?? ()                                        No symbol table info available.                                #6  0xf631ffff in ?? ()                                        No symbol table info available.                                #7  0x00c6e946 in ?? ()                                        No symbol table info available.                                #8  0x3c890000 in ?? ()                                        No symbol table info available.                                #9  0x8e4ae824 in ?? ()                                        No symbol table info available.                                #10 0xc689fffe in ?? ()                                        No symbol table info available.                                #11 0x2824448b in ?? ()                                        No symbol table info available.                                #12 0x0c74c085 in ?? ()                                        No symbol table info available.                                #13 0xe8240489 in ?? ()                                        No symbol table info available.                                #14 0xfffe4f7e in ?? ()                                        No symbol table info available.                                Backtrace stopped: previous frame inner to this frame (corrupt stack?)
```

dpgeorge, 2017-09-06:
> Though mp-cross is compiled with -g it has no symbol table!

Try explicitly compiling and running mpy-cross:
```
$ cd mpy-cross
$ make clean
$ make DEBUG=1
$ ./mpy-cross test.py
```
Try that with and without a valid test.py file (just put some dummy Python code in test.py).

ASMfreaK, 2017-09-06:
@dpgeorge
Compiling and running inside mpy-cross works
The issue is in asm block in nlr_push (nlr=0x0) at ../py/nlrx86.c:53
See full compilation and debug log in [this gist](https://gist.github.com/ASMfreaK/48e71bd9d20c3ddf1104ec25edaf8782). 
I'm not sure but it may be caused by different calling conventions on Android. [Maybe some info from here will help](https://developer.android.com/ndk/guides/abis.html#86-64). Though it seems that android x86 ABI should be no diffent from normal.

dpgeorge, 2017-09-06:
@ASMfreaK can you please provide a full disasm of the mpy-cross executable?  Something like `objdump mpy-cross`.

ASMfreaK, 2017-09-06:
@dpgeorge
[Output of `objdum -d ./mpy-cross`](
https://gist.github.com/469d0daadc298be670e03846b44d2c10)

dpgeorge, 2017-09-06:
It looks like the ABI is different... try removing line 59 of py/nlrx86.c, the one that has "pop %ebp" on it.

ASMfreaK, 2017-09-06:
It works! I've enhanced [the change you proposed](https://gist.github.com/ab55345a0fd5f088a49df8f693858334).
Micropython passes all tests via
`MICROPY_MICROPYTHON=../ports/unix/micropython ./run-tests`
Surprisingly bench-tests fails:
`MICROPY_MICROPYTHON=../ports/unix/micropython ./run-bench-tests`
with this error:
```
bench/arrayop:                                              

Traceback (most recent call last):                            

File "bench/arrayop-1-list_inplace.py", line 4, in <module>                                                             

File "/data/data/com.termux/files/home/projects/micropython/tests/bench/bench.py", line 1, in <module>                

ImportError: no module named 'time'                         
Traceback (most recent call last):                            

File "./run-bench-tests", line 97, in <module>                main()                                                    

File "./run-bench-tests", line 93, in main                    
if not run_tests(pyb, test_dict):                         

File "./run-bench-tests", line 44, in run_tests               
output_mupy = float(output_mupy.strip())
ValueError: could not convert string to float: 
b'CRASH'
```
Can this be caused by the incompatible (3.6) version of default python

dpgeorge, 2017-09-08:
Great that it works.  But I wonder if it's just because you use clang as the compiler, in which case the macro condition needs to include a check for that.

@pfalcon can you check if the current code runs on Android, from the tests you did some time ago?  Back then the code had nlrx86.S written in assembler and so the move to nlrx86.c written in C may have broken Android for all compilers.

pfalcon, 2017-09-08:
> @pfalcon can you check if the current code runs on Android

When I was working on Android port, I definitely tried it only on ARM systems. I can't test it soon, it's too big a recursion from current things I'm working on.

> But I wonder if it's just because you use clang as the compiler

I doubt that, it seems it's a kind of "new" x86 ABI, used where legacy constraints don't apply (Zephyr, Android, etc.) So, my suggestion is:

@ASMfreaK : Please consider submitting a patch, otherwise it may be lost and next guy like you will need to redo what you've done again.

ASMfreaK, 2017-09-08:
@dpgeorge
I'll try to test compilation on ARM Android on termux. I wonder if there will be any problems.

ASMfreaK, 2017-09-08:
@dpgeorge I've tested compiling unix port on android arm. This time I got [these errors](https://gist.github.com/07ee95fbdb5cb0eb5c51c0fd4b0b5ec0)

dhylands, 2017-09-09:
The first line of your output reports:
```
make: pkg-config: Command not found
```
You should install pkg-config, do a clean, and try again. The asm errors you're seeing are most likely not related to this, but you'll most likely encounter errors during the linking stage if pkg-config isn't installed.

ASMfreaK, 2017-09-09:
@dhylands 
I've installed pkg-config, but the asm error didn't go away as you said.

dpgeorge, 2019-07-30:
The x86 NLR code was improved in 52620c6b0e8a601e49866d0db37abbe217b16a0d and c97607db5ccc03afbccacf853f2cd06305c28251.  The Thumb NLR code was improved in 97cc48553828ed091325d0d3922eb4cd6c377221 and deaa46aa66230071d355c0ffd004ccf0cbefd8b9.  Hopefully these fix the above build errors.  Please reopen the issue if they don't.

closed 2019-07-30 by dpgeorge