micropython/micropython #3291: Classifying unary and binary operations properly

pfalcon, 2017-09-02:
In a weird twist of interest, I'd like to have another stab at implementing fuller Python data model, even if off by default (enabled only for coverage build). One topic in this is making sure we can support full inventory of special methods. And to show off the problem, a patch like:

~~~
--- a/py/runtime0.h
+++ b/py/runtime0.h
@@ -49,55 +49,74 @@ typedef enum {
     MP_UNARY_OP_POSITIVE,
     MP_UNARY_OP_NEGATIVE,
     MP_UNARY_OP_INVERT,
+    MP_UNARY_OP_ABS, // __abs__
     MP_UNARY_OP_NOT,
     MP_UNARY_OP_SIZEOF, // for sys.getsizeof()
 } mp_unary_op_t;
~~~

breaks the stuff (weird errors in the testsuite of not supported *binary* ops). I traced that to:

~~~
# vmentrytable.h
    [MP_BC_UNARY_OP_MULTI ... MP_BC_UNARY_OP_MULTI + 6] = &&entry_MP_BC_UNARY_OP_MULTI,
    [MP_BC_BINARY_OP_MULTI ... MP_BC_BINARY_OP_MULTI + 35] = &&entry_MP_BC_BINARY_OP_MULTI,
~~~

So, vmentrytable.h silently assumes there're 7 bytecod'able unary ops, and adding one inbetween then causes overlap with with binary ops. So, we definitely should add cross-referencing comments about such code allocations, if not static asserts or similar. And it's just another reason to at least encode complete inventory of operations (if not completely implement them).

But the issue is even deeper. `__abs__` unary op doesn't need to be encoded as a bytecode, because corresponding operation is triggered by a function, not an operator. Thus, the sole purpose of this unary op is to correspond to `__abs__` special method. But then suddenly the same would apply to `__hash__` and `__len__`. Unwinding this further, the unary operations can be classified as:

1. Ones which appear in the bytecode (vs not).
2. Ones which correspond to the special methods (vs not).

These are essentially partial criteria, which enforce partial order on the operations. And by merging these together in a total ordering, smartly, we can optimize various implementation aspects, e.g. not allocate bytecodes to ops which can't appear in it, and not waste space in op -> specmeth name tables for ops which don't have associated special methods.

It gets even more complicated for binary ops, the can be:

1. Ones which appear in the bytecode (vs not).
2. Ones which correspond to the special methods (vs not).
3. Ones which have inplace and reverse counterpart.

E.g. DIVMOD doesn't have inplace/reverse. EXCEPTION_MATCH doesn't correspond to a special meth Reverse ops never appear in the bytecode.

From this classification of binary ops, following constraints can be deduced which are again could be implemented by smart total ordering on them:

a) ops appearing in the bytecode should be allocated bytecode codes, others - not
b) ops corresponding to spec meths should be efficiently convertible to meth names, others - don't need to be
c) inplace ops should be easily convertible to normal, normal to reverse, and vice versa.

The adhoc workaround for MP_UNARY_OP_ABS is easy - just move it after NOT, but my adhoc attempts to reorder binary ops to support efficient implementation of inplace and reverse operations so far failed (leads to weird binary op failures in the testsuite again). That's why I propose to fix this issue in thorough manner, applying suitable optimizations and documenting this things, not just "making it work".

pfalcon, 2017-09-05:
> Feel free to work on something in this direction. I don't have any ideas or implementations at my end.

As I mentioned above, I tried, and hit weird breakage, so will need your help to figure it out.

Specific proposals are coming next.

pfalcon, 2017-09-05:
So, I posted https://github.com/micropython/micropython/pull/3300 to support __rOP__ special methods. As the comment in that PR says:

> // TODO: If OP_* and OP_REVERSE_* ranges are adjacent, this if is not needed.

So, these ranges should be adjacent for efficiency. At the same time, bytecode opcodes should not be allocated for `OP_REVERSE_*`. It's logical to make first half of operations in mp_binary_op_t to be allocatable in bytecode, and 2nd half - not. Then, `OP_*` should go first, immediately followed by `OP_REVERSE_*`.

But it also makes sense to make `OP_INPLACE_*` to be adjacent too, but then the only place for it is before `OP_*`.

And we talk about `OP_*`, it's the range which has direct correspondence to `OP_INPLACE_*` and `OP_REVERSE_*` (12 operations).

`OP_DIVMOD` is foreign in this regard, nor it is used in bytecode. Then the logical place for it is after `OP_REVERSE_*`.

What's left are relational operators, the only place for them then is the very beginning of mp_binary_op_t. (Except for those ops among them which can't appear in bytecode, which should go to the very end of it, but let's leave that for the later.)

But, as I mentioned above, when I tried to reorder ops like that, I was getting weird testsuite failures.

pfalcon, 2017-09-05:
Actually, let's start with a simple one:

~~~
--- a/py/runtime0.h
+++ b/py/runtime0.h
@@ -68,7 +68,6 @@ typedef enum {
 
     MP_BINARY_OP_MODULO,
     MP_BINARY_OP_POWER,
-    MP_BINARY_OP_DIVMOD, // not emitted by the compiler but supported by the runtime
     MP_BINARY_OP_INPLACE_OR,
     MP_BINARY_OP_INPLACE_XOR,
 
@@ -98,6 +97,8 @@ typedef enum {
     // these are not supported by the runtime and must be synthesised by the emitter
     MP_BINARY_OP_NOT_IN,
     MP_BINARY_OP_IS_NOT,
+
+    MP_BINARY_OP_DIVMOD, // not emitted by the compiler but supported by the runtime
 } mp_binary_op_t;
~~~

i.e., moving MP_BINARY_OP_DIVMOD at the end of mp_binary_op_t. This leads to failure of builtin_divmod_intbig:

~~~
Traceback (most recent call last):
  File "builtin_divmod_intbig.py", line 6, in <module>
  File "builtin_divmod_intbig.py", line 4, in <module>
TypeError: unsupported types for : 'int', 'int'
~~~

Why?

pfalcon, 2017-09-05:
> Why?

A: Because of:

~~~
     } else if (op <= MP_BINARY_OP_INPLACE_POWER) {
~~~

So, let's do it step by step:

1. Please approve moving OP_INPLACE ahead of OP.
2. Please approve moving DIVMOD at the very end.

And let's propagate changes required for that.

dpgeorge, 2017-09-06:
> A: Because of:
> 
>      } else if (op <= MP_BINARY_OP_INPLACE_POWER) {

It would be good to replace that check with a macro, like `MP_BINARY_OP_CHECK_ARITH(op)`, and define that alongside the enum in runtime0.h.  That way it's much easier to see what relies on the ordering and the macro can be changed if needed.

The other place where binary op ordering is somewhat important is in parse.c, in fold_constants().  But that just relies on the ordering within the 12 distinct arithmetic operations and doesn't care if this block of 12 shifts around.

> Please approve moving OP_INPLACE ahead of OP.

That sounds ok.  Only reason it is where it is currently is that non-inplace ops seem more fundamental than inplace and so non-inplace went first.

> Please approve moving DIVMOD at the very end.

Also sounds ok.  The reason it's there is because it groups more logically with the non-inplace ops, but it can really go anywhere in the enum definition.

pfalcon, 2017-09-06:
> It would be good to replace that check with a macro, like MP_BINARY_OP_CHECK_ARITH(op)

Yep, maybe not on iteration 1, but would be good.

> The other place where binary op ordering is somewhat important is ...

I'm sure we have more places like that, that's I'd like to do it iteratively, because otherwise it's long time to prepare and potentially big noisy patch to push.

So, to clarify, I'm going to do it step by step, potentially pushing each step directly (well, will try to buffer up/squash things of course).

pfalcon, 2017-09-07:
Started step by step refactorings in d4d1c45a553c6361a72053383f0fe242f05cad3d and following commits.

pfalcon, 2017-09-09:
b8ee7ab5b988f424e34e778fe3a679710d675efa marks completion of the 1st stage of the refactor, after which efficient support for reverse ops methods can be added.

There're still more other steps to do.

pfalcon, 2017-09-17:
In 9dce823cfd0a9991350184f08a1373f3887134f4, MP_UNARY_OP_ABS is introduce and abs() implementation refactored to use it, leading to code size reduction (few issues were fixed on the way too).

pfalcon, 2017-10-21:
Next step: https://github.com/micropython/micropython/pull/3377

Remaining step: Implement MICROPY_PY_ALL_SPECIAL_METHODS faithfully.

pfalcon, 2017-10-21:
After that, the only not implemented feature will be from this note https://docs.python.org/3/reference/datamodel.html#object.__ror__ :

> If the right operand’s type is a subclass of the left operand’s type and that subclass provides the reflected method for the operation, this method will be called before the left operand’s non-reflected method. This behavior allows subclasses to override their ancestors’ operations.

pfalcon, 2017-10-21:
(datetime implementation from CPython relies on it for example.)

dpgeorge, 2017-10-24:
That subclassing rule is a tricky one...

Also note that `__divmod__` and `__rdivmod__` are part of the list of supported operators in CPython, but in uPy the divmod is not a bytecode and doesn't fit within the 12 ops that uPy supports.  So if uPy wanted to support `__rdivmod__` then either it must be moved to within the exist 12 ops (hence changing the bytecode) or there needs to be some special case in the reversing code to handle divmod.

pfalcon, 2017-10-25:
> So if uPy wanted to support __rdivmod__

Well, I guess we can live without it for now, just the same as w/o matrix mul for example...

pfalcon, 2017-11-01:
> That subclassing rule is a tricky one...

Apparently it's tricky if people still try to fix it: https://github.com/python/cpython/pull/1325/files

jonnor, 2024-09-29:
The referenced PR seems to have been merged, and I see a wide range of unary and binary operations now in py/runtime0.h - including the ones mentioned here as needing to be added - like MP_UNARY_OP_ABS and MP_BINARY_OP_DIVMOD.
So this seems to largely have been fixed/implemented?

closed 2025-02-04 by projectgus