micropython/micropython #3293: Generator function has no __name__ attribute

richarddd, 2017-09-02:
Im in a situation when im required to save the name of the function from a decorator:

```
# ugly
def _hack_generator_name(fn):
    if not inspect.isgeneratorfunction(fn):
        return fn.__name__
    try:
        fn()
    except Exception as e:
        name = str(e)
        return name[:name.index("(")]

def optional_arg_decorator(fn):
    def wrapped_decorator(*args):
        if len(args) == 1 and callable(args[0]):
            return fn(args[0])
        else:
            def real_decorator(decoratee):
                return fn(decoratee, *args)
            return real_decorator

    return wrapped_decorator

@optional_arg_decorator
def httppost(fn, path=None, name=None):
    _decorator_list.append((_hack_generator_name(fn), path, Request.POST))
    return fn

@optional_arg_decorator
def httpget(fn, path=None, name=None):
    _decorator_list.append((_hack_generator_name(fn), path, Request.GET))
    return fn
```

But fn.__name__ is missing from generator functions.

I solve this issue temporary by using an ugly hack (by calling the method and finding the name from the exception...)

dpgeorge, 2017-09-05:
It's a bit of a shame that there is code to print the generator name but not extract it in a programatic way (ie via `.__name__`).  The feature could be added almost for free by reusing the fun_bc_attr() function for normal functions, but there's one level of extra indirection that's needed to extract the name from a generator....

dpgeorge, 2018-07-10:
In b488a4a8480533a6a3c9468c2f8bd359c94d4d02 generating functions were optimised so they used the same structure as normal bytecode functions.  This made it possible to add support for `__name__` to generating functions very cheaply.  See e2e22e3d7e29985579b2c91c639c71229422f349

I think this now fixes the issue here.

closed 2018-08-13 by dpgeorge