micropython/micropython #3301: Port to run on Texas Instruments CC2538

ricardovf, 2017-09-05:
Hi, first, congratulations on the project, its really great. I am doing a seminar on the micropython and im just starting to study the codebase. Would be great to know how hard it would be to port the code to run on the CC2538. I saw its a ARM Cortex M3, whereas pyboard is Cortex M4, and there is not much diference between them (FPU and DSP on M4).

Do you have any ideia? I would like to have an estimate so i can try to do it for my seminar.

Thanks!

dpgeorge, 2017-09-05:
> congratulations on the project, its really great

Thank you!

> Would be great to know how hard it would be to port the code to run on the CC2538. I saw its a ARM Cortex M3, whereas pyboard is Cortex M4, and there is not much diference between them

The Cortex M3 and M4 are very similar, and the M3 is already supported by MicroPython.  The more difficult thing for any bare-metal port is to get the MCU up and running, with clock settings, IRQ table, and then getting a UART working (to make the REPL).  Beyond that, for the CC2538 you'd want to support the wireless peripherals and that is the thing that will take most of the time and effort.  It looks like it has just 32k of RAM and that is enough although not much if you need to dedicate some of it to the wireless stacks.

A better place for this discussion is https://forum.micropython.org, so please head over there if you want to continue with this idea (or anything else related to MicroPython).

closed 2017-09-05 by dpgeorge

ricardovf, 2017-09-05:
Thanks for the fast answer Damien. I will talk to my teacher and if we decide to try to port, i will go to the forum.