micropython/micropython #3302: PYBv10: udp socket in the multi-threaded environment blocks the running of micropython

LiWeiwei-GitHub, 2017-09-06:
I used w5200 in a multi-threaded environment to write a udp socket program.   

UDP socket default is blocking mode. I thought in a multi-threaded environment, the socket will only block the caller of **recvfrom**, and other threads are not affected, but that's not the case.  

I do not know if this should be the case on PYBv10, please tell me if someone has experience.  

The following is the test code:
```
# -*- coding: utf-8 -*-

import pyb
import network
import socket
import _thread

def work_thread():
    while True:
        print('work_thread')
        busy_delay = 1000
        while busy_delay:
            busy_delay = busy_delay - 1

def sock_thread(sock):
    while True:
        print('sock_thread')
        data, addr = sock.recvfrom(1024)

def test():
    nic = network.WIZNET5K(pyb.SPI(1), pyb.Pin.board.X5, pyb.Pin.board.X4)
    nic.ifconfig(('192.168.1.2', '255.255.255.0', '192.168.1.1', '192.168.1.1'))
    print(nic.ifconfig())

    _thread.start_new_thread(work_thread, ())

    sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
    sock.bind(('', 1031))

    _thread.start_new_thread(sock_thread, (sock,))

    pyb.delay(24 * 60 * 60 * 1000)

```
The results of the running are as follows:
```
MicroPython v1.9.2-58-g069fc48-dirty on 2017-09-06; PYBv1.0 with STM32F405RG
Type "help()" for more information.
>>> import sock_test
>>> sock_test.test()
('192.168.1.2', '255.255.255.0', '192.168.1.1', '192.168.1.1')
work_thread
work_thread
work_thread
work_thread
sock_thread

```
After the call of **sock.recvfrom(1024)** in **sock_thread**, the **work_thread** does not running any more.

dpgeorge, 2017-09-06:
The issue is probably that the GIL is not being released around the recvfrom call.  If you are able to edit the source and recompile, you can try adding MP_THREAD_GIL_EXIT/MP_THREAD_GIL_ENTER around the call to recvfrom in modnwwiznet5k.c, something like:
```diff
--- a/ports/stm32/modnwwiznet5k.c
+++ b/ports/stm32/modnwwiznet5k.c
@@ -265,7 +265,9 @@ STATIC mp_uint_t wiznet5k_socket_sendto(mod_network_socket_obj_t *socket, const
 
 STATIC mp_uint_t wiznet5k_socket_recvfrom(mod_network_socket_obj_t *socket, byte *buf, mp_uint_t len, byte *ip, mp_uint_t *port, int *_errno) {
     uint16_t port2;
+    MP_THREAD_GIL_EXIT();
     mp_int_t ret = WIZCHIP_EXPORT(recvfrom)(socket->u_param.fileno, buf, len, ip, &port2);
+    MP_THREAD_GIL_ENTER();
     *port = port2;
     if (ret < 0) {
         wiznet5k_socket_close(socket);
```

LiWeiwei-GitHub, 2017-09-06:
I test it again, it works fine. Thanks.  
The **wiznet5k_socket_recv** in **modnwwiznet5k.c** maybe have the same problem. @dpgeorge

dpgeorge, 2017-09-06:
Ok, great, thanks for testing!  Yes, some of the other functions will need this patch.

dpgeorge, 2017-09-08:
Should be fixed by cc7fece309b0ce6d361cade8690b6c3a162d7378

closed 2017-09-08 by dpgeorge