micropython/micropython #3305: math.acos() returns NaN instead of raising a ValueError

deshipu, 2017-09-07:
Under CPython:
```
>>> math.acos(-2)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: math domain error
```

Under MicroPython:
```
>>> math.acos(-2)
nan
```

Seems like MicroPython is turning into PHP more and more. Returning a NaN on error is a very bad idea, because it complicates error handling and leads to a lot boilerplate code, but also because it lets an error propagate through the program (NaN being "contagious" in the sense that any operation involving it results in NaN) and makes it extremely hard to track down the actual source of the error.

stinos, 2017-09-07:
> turning into PHP more and more

Just curious, did you assess this based on just this issue, or are there other ones which in your opinion contribute to that?

deshipu, 2017-09-07:
There are more places where errors are silently ignored or a wrong value is returned, but I'm not going to discuss it here. It was a mistake to include that remark in the bug report in the first place, I apologize for that. Please disregard it.

dpgeorge, 2017-09-08:
There are quite a few math (and cmath) operations that don't correctly check for domain errors.  It's on my list to fix these but likely it would be an optional check, something like MICROPY_SAFE_MATH.

deshipu, 2017-09-08:
Would a pull request with that be useful? I'm asking, because I don't want to do the work just to have it again rejected on the grounds of increasing the binary size or worsening performance.

dpgeorge, 2017-09-26:
See #3331 for a PR that does the checking.

dpgeorge, 2017-10-10:
Now fixed, see above PR.

closed 2017-10-10 by dpgeorge