micropython/micropython #3314: ustruct unpack_from with nonzero offset doesn't match CPython  (esp8266)

eliotb, 2017-09-12:
Unpacking an integer at a byte offset not a multiple of integer size gives unexpected result. when explicit endian indication is not given.

`struct.unpack_from('i', buf, 1)` returns integer at offset 4

Test code:
```python

try:
    import ustruct as s
except ImportError:
    import struct as s

def as_hex(a):
    return str(tuple((hex(x) for x in a)))

def peval(v):
    print(v, '=', as_hex(eval(v)))

b = bytearray(range(1,13))

peval("s.unpack('i', b[0:4])")
peval("s.unpack_from('i', b, 0)")

peval("s.unpack('i', b[1:5])")
peval("s.unpack_from('i', b, 1)")

peval("s.unpack('bi', b[0:8])")
peval("s.unpack_from('bi', b, 0)")

peval("s.unpack('<bi', b[0:5])")
peval("s.unpack_from('<bi', b, 0)")
 
```
**cpython3:**
s.unpack('i', b[0:4]) = ('0x4030201',)
s.unpack_from('i', b, 0) = ('0x4030201',)

s.unpack('i', b[1:5]) = ('0x5040302',)
s.unpack_from('i', b, 1) = ('0x5040302',) <-- reference
s.unpack_from('<i', b, 1) = ('0x5040302',)

s.unpack('bi', b[0:8]) = ('0x1', '0x8070605')
s.unpack_from('bi', b, 0) = ('0x1', '0x8070605')

s.unpack('<bi', b[0:5]) = ('0x1', '0x5040302')
s.unpack_from('<bi', b, 0) = ('0x1', '0x5040302')

**micropython:**
s.unpack('i', b[0:4]) = ('0x4030201',)
s.unpack_from('i', b, 0) = ('0x4030201',)

s.unpack('i', b[1:5]) = ('0x5040302',)
s.unpack_from('i', b, 1) = ('0x8070605',) <-- Incorrect?
s.unpack_from('<i', b, 1) = ('0x5040302',)

s.unpack('bi', b[0:8]) = ('0x1', '0x8070605')
s.unpack_from('bi', b, 0) = ('0x1', '0x8070605')

s.unpack('<bi', b[0:5]) = ('0x1', '0x5040302')
s.unpack_from('<bi', b, 0) = ('0x1', '0x5040302')

dpgeorge, 2017-09-12:
If you don't specify a byte ordering (ie one of `<>=@!`) then `@` is used by default, which is native ordering and native alignment.  In native mode, uPy is aligning the first data value that is packed or unpacked, whereas CPython is not.  The CPython docs explicitly mention this in the notes of https://docs.python.org/3/library/struct.html#byte-order-size-and-alignment :

> Padding is only automatically added between successive structure members. No padding is added at the beginning or the end of the encoded struct.

So in this respect uPy is non-compliant.

dpgeorge, 2017-09-19:
The reason no padding is added at the beginning or end of a struct is so that the value returned from `calcsize()` is the actual number of bytes that will be used by `pack()` and `unpack()`.  `calcsize` assumes that the first member is already aligned.

dpgeorge, 2019-07-01:
A related issue: `struct.unpack('h', memoryview(b'\x01\x02\x03')[1:])` in CPython gives `(770,)` but in MicroPython it gives `(3,)`

dpgeorge, 2019-08-05:
This incorrect behaviour shouldn't be let through silently.  Either 1) it should detect these cases and raise an exception; 2) it should handle these cases correctly.

dpgeorge, 2019-09-02:
Fixed by 1022f9cc35564b216a4bcd7c65e8243c810a0ca9 (and `struct.pack_into` fixed by 24c3e9b283da26093ca653fc6b441042fedec135)

closed 2019-09-02 by dpgeorge