micropython/micropython #3317: wipy(esp32) Timer

L-eeX, 2017-09-15:
Hi,just play with esp32(wipy),in the terminal,input:
 >>> from machine import Timer
 >>> tim=Timer(-1)
 >>> tim
 I get->
 >>> Timer(3ffv4c80; alarm_en=0, auto_reload=1, counter_en=0)

about this messge,how I can understand the Timer(...) means about?thanks a lot.

peterhinch, 2017-09-15:
You'll get a better response if you raise this in [the forum](https://forum.micropython.org/). This is the MicroPython bug tracker.

pfalcon, 2017-09-16:
Thanks for submission to the MicroPython bug tracker. This bug tracker is intended to track confirmed bugs in the MicroPython codebase. However, your submission doesn't look like a bugreport. For questions, support requests, suggestions, etc. you may find that using the forum gives better result:

forum.micropython.org

You can get even better results by first searching for existing topic where a similar issue may be discussed (this is also a good etiquette on any Internet forum, as otherwise you may end up asking a question which was answered just a day or an hour ago).

closed 2017-09-16 by pfalcon