micropython/micropython #3319: Ideas on code size optimization

pfalcon, 2017-09-16:
This ticket is intended to collect ideas on how to optimize for code sizes, which aren't carried out immediately for some reason (e.g. they make a breaking change, required big start-up effort, or complicate maintenance).

pfalcon, 2017-09-16:
Switch btree.open() to accept cache size in pages, with removal of bytes-to-pages conversion from BerkeleyDB fork: https://github.com/micropython/micropython/issues/3257#issuecomment-328335244 . Saves 54 bytes on Xtensa (esp8266).

pfalcon, 2017-10-30:
Remove uio.open(), nobody uses it anyway.

pfalcon, 2017-10-30:
Remove ure.DEBUG flag from ure (and assocaited re1.5 compiled re dump code).

pfalcon, 2017-10-30:
Remove module-level ure.match() and ure.search().

dpgeorge, 2017-10-31:
> Remove module-level ure.match() and ure.search().

-1, they are very useful.

robert-hh, 2017-10-31:
-1, as this is present since long and may break existing code. 

> Remove ure.DEBUG flag from ure (and assocaited re1.5 compiled re dump code).

ok: This should not break existing code, and it's output is anyhow hard to understand

> Remove uio.open(), **nobody uses it anyway**.

True for me, but how do you know? Is there a bug in it and nobody yet complained?

pfalcon, 2017-10-31:
> -1, they are very useful.

And yet that's the way to get smaller binary code size without losing functionality. This ticket is a raw dump of ideas of different "levels", ideally, all of them implemented, definitely, all implemented in a configurable way, and values for configuration shipped in mainline are different of course.

stinos, 2017-10-31:
> Remove module-level ure.match() and ure.search().

also -1 unless they get added in micropython-lib or so: these are like *the* go-to functions for easy working with regexes

dpgeorge, 2022-03-30:
`ure.DEBUG` was made optional in 7d851a27f146188752e89bb026021fb8d3985395

Otherwise, I'll close this issue because it's quite stale.

closed 2022-03-30 by dpgeorge