micropython/micropython #3323: extmod/modframebuf.c Instantiating a subclass of FrameBuffer produces error.

peterhinch, 2017-09-20:
Tested on MicroPython v1.9.2-87-gda8c4c2-dirty on 2017-09-15; PYBv1.0 with STM32F405RG
```python
from framebuf import FrameBuffer, MVLSB

class FOO(FrameBuffer):
    def __init__(self):
        height = 64
        width = 128
        self.buffer = bytearray(height * width // 8)
        super().__init__(self.buffer, width, height, MVLSB)

foo = FOO()
```
Outcome:
```
Traceback (most recent call last):
  File "<stdin>", line 10, in <module>
TypeError: function missing 4 required positional arguments
```

dpgeorge, 2017-09-20:
Using native classes as a base class is only partially supported in uPy, and you've hit on the case where the base class (FrameBuffer) doesn't support construction with zero arguments.

The following may seem like a bit of magic but it will allow you to do what you want:
```python
import framebuf
class FNative(framebuf.FrameBuffer):
    pass
class F(framebuf.FrameBuffer):
    def __new__(cls, w, h):
        print('new', w, h)
        self = FNative(bytearray(w * h * 2), w, h, framebuf.RGB565)
        self.member = 1234
        return self
f = F(3, 3)
print(f.member)
f.fill(0x11)
f.hline(0, 0, 3, 0x2345)
print(bytes(f))
```

peterhinch, 2017-09-20:
So do you want me to adopt this approach with PR #3318? My solution lacks magic but is arguably more easily understood :)

dpgeorge, 2017-09-21:
> So do you want me to adopt this approach with PR #3318?

If it works, the way here is more efficient in terms of RAM usage so would be preferred.  Also, the magic part of it is just a temporary (hopefully!) work-around and will eventually be replaced with the proper construct.

peterhinch, 2017-09-21:
On reflection I feel queasy about taking responsibility for code whose magic I don't actually understand. Perhaps someone more knowledgeable should take over.

pfalcon, 2017-09-21:
@peterhinch , fair enough, and I'm +1 for *not* going where @dpgeorge suggests. But then feel free to reflect on the comment https://github.com/micropython/micropython/pull/3318#issuecomment-329961998 - to somebody, *your* code may be already not easy to understand (without a comment).

peterhinch, 2017-09-22:
@pfalcon I do plan to add a comment if my approach is adopted.

dpgeorge, 2017-09-22:
> On reflection I feel queasy about taking responsibility for code whose magic I don't actually understand.

Ok, then please just do it as you already have it.  Please then make the comment point out that inheritance would be the idea solution but it's not fully supported.  Thanks!

peterhinch, 2017-09-22:
@dpgeorge I made an attempt to make this work but without success - doubtless because I lack the thaumaturgy. I'll have to pass the ball on this one.

dpgeorge, 2017-09-22:
> I'll have to pass the ball on this one.

But I said above it was OK to keep #3318 as it is (with comments added)...

peterhinch, 2017-09-22:
@dpgeorge OK, now done.

dpgeorge, 2017-12-12:
Ok, this issue should now be fixed by d32d22dfd79439e07739166eaa3f7e41466c7ec8.  An explicit test for subclassing framebuf.FrameBuffer was added in 3c28df16586157b6b80e7437559c36f05a309e24

closed 2017-12-12 by dpgeorge

peterhinch, 2017-12-12:
It does. I have raised a PR #3485.

peterhinch, 2017-12-22:
Unfortunately I made a mistake when testing this, forgetting that the official driver was frozen as bytecode. I was therefore testing the release code rather than my patch. This became apparent when I spotted [this typo](https://github.com/micropython/micropython/blob/9bcdb0acd1ea059533ba01d8c8ccfd12adeb0eee/drivers/display/ssd1306.py#L35). After fixing this the code produced errors. I reduced it down to the following test case:
```python
import framebuf

class Foo(framebuf.FrameBuffer):
    def __init__(self):
        pages = 24 // 8
        buffer = bytearray(pages * 128)
        super().__init__(buffer, 128, 24, framebuf.MONO_VLSB)

class Bar(Foo):
    def __init__(self):
        super().__init__()

class Rats(Bar):
    def __init__(self):
        super().__init__()

rats = Rats()
```
When pasted it produces the following outcome
```
Traceback (most recent call last):
  File "<stdin>", line 17, in <module>
TypeError: function missing 4 required positional arguments
```
It seems that the `FrameBuffer` constructor is being called in place of that for `Rats`.

Test conditions MicroPython v1.9.3-117-g48dc826-dirty on 2017-12-22; PYBv1.0 with STM32F405RG

Apologies for my error in testing this.

peterhinch, 2018-01-03:
Please ignore the above - I must have been suffering from brain-fade in the runup to Christmas and tested against an incorrectly built firmware image.
See #3535 which fixes this driver.