micropython/micropython #3324: implicit declaration of function mp_hal_set_interrupt_char

microwavesafe, 2017-09-21:
This is only a minor build error (it still builds, but it would be nice to clean it up). In a custom port with MICROPY_KBD_EXCEPTION turned on, I am getting two implicit declaration errors, when calling function mp_hal_set_interrupt_char(). One in modmicropython.c and the other in pyexec.c.

I was about to submit a pull request, but I wanted to check if adding
```
#include "lib/utils/interrupt_char.h"
```
to the top of modmicropython.c would be the thing to do. I can add the include inside the MICROPY_KBD_EXCEPTION define to not pull it in unless required, but I can't see any problem with just including it anyway.

Adding it to the top of pyexec.c looks OK to me as it is in the same folder.

dpgeorge, 2017-09-22:
The way that the ports deal with this is by defining `mp_hal_set_interrupt_char()` in their mphalport.h file, which can be as simple as including `lib/utils/interrupt_char.h`.  This way you can make it inline, a macro, etc.

microwavesafe, 2017-09-22:
OK, thanks for the help.

closed 2017-09-22 by microwavesafe