micropython/micropython #3325: openwrt/lede build error

apatrushev, 2017-09-22:
Latest `micropython` failed to build for `openwrt`/`lede`. The reason is strict check here: https://github.com/micropython/micropython/blob/72732fea1a041b94448d98ad4fd53545caa758ab/py/persistentcode.c#L377

Not all openwrt/lede target platforms are `__arm__`. I suggest to remove `__arm__` check from ifdef.

pfalcon, 2017-09-22:
@dpgeorge : Why do we have mpy-cross specific code in the core in the first place?

dpgeorge, 2017-09-24:
@apatrushev I assume the build is failing when building the "mpy-cross" component, right?

> Why do we have mpy-cross specific code in the core in the first place?

That code is supposed to be "generic", useful for any port to save bytecode, and was there before mpy-cross existed; see d8c834c95d506db979ec871417de90b7951edc30

As the comment says, this code needs to be abstracted away properly by ports providing "save to file" functionality.  But to fix it temporarily this code could be wrapped by a new config variable such as MICROPY_PERSISTENT_CODE_SAVE_POSIX (since it relies on open, write and close functions), or an easier thing is to just do what's done by #3326.

dpgeorge, 2017-09-25:
Fixed by PR linked above.

closed 2017-09-25 by dpgeorge