micropython/micropython #3327: Add EINTR to uerrno module?

dlech, 2017-09-22:
This error is common on Unix platforms. Just wanted to ask if it would be worth adding to the uerrno module before submitting a pull request. `EINTR` already is available via the errno module in micropython-lib.

pfalcon, 2017-09-23:
There're ~200 errors common on Unix platforms. Please read https://github.com/micropython/micropython/wiki/ContributorGuidelines and feel free to provide arguments why EINTR is unavoidable useful (especially that in upstream Python 3.6 users would never see it).

dlech, 2017-09-23:
Since pretty much all system calls can be interrupted by a signal which results in this error, it seems more common than the others. Since I can use errno micropython-lib, this isn't a blocker for me, but I just wanted to through it out there to see if anyone else had a need too. If I'm the only one, then there isn't really a compelling reason to add it.

pfalcon, 2017-09-23:
> Since I can use errno micropython-lib

That's the recommended way indeed.

dpgeorge, 2017-09-25:
Ideally EINTR wouldn't filter through because of PEP 475 (which isn't yet implemented in uPy).  See #1329 and #3268 for further reading.

dlech, 2017-09-25:
Thanks. In that case, I'm closing in favor of #3268 / PEP 475.

closed 2017-09-25 by dlech

dlech, 2020-03-05:
PEP 475 implemented in #5723