micropython/micropython #3334: Making string from bytes sometimes results in (null)

jpochyla, 2017-09-27:
```python
>>> print(str(bytes(), 'utf-8'))
(null)
```

The problem seems to be here: https://github.com/micropython/micropython/blob/master/py/objstr.c#L158-L172, `data` pointer is directly copied from `mp_const_empty_bytes_obj`, which is `NULL`. I'm not sure if and how to fix the issue.

dpgeorge, 2017-09-28:
Nice find, thanks for the report.

There are a few ways to fix it:
1. make `mp_const_empty_bytes_obj` point to the empty string "" instead of NULL.
2. make `mp_obj_str_make_new` explicitly check for NULL data pointers.
3. fix `mp_vprintf` so it doesn't print `(null)` for the case where it's passed a zero-length string via the `"%.*s"` format spec.

The last one should anyway be fixed, which would solve the issue.  But I wonder if there are any other edge cases which would arise because of the NULL pointer in the empty bytes object... on the surface it looks like there could be because some routines (eg string formatting code) expect all strings and bytes objects to be terminated by a null byte (and so the empty bytes object should have a pointer to an empty ASCIIZ string).  But all the code I looked that assumes null terminated strings is safe because when the string has zero length it terminates early.

jpochyla, 2017-10-02:
I think a combination of 1 and 3 seems like the best solution.

Ad 1), from a perspective of extmod development, it was surprising to discover that `mp_buffer_info_t.buf` can also be NULL (event though it's documented in the code). Should we also expect to receive 0-terminated str here?

pfalcon, 2017-10-02:
I guess code size should drive a solution here, and we likely have an empty string in the codebase already, so 1) should be both free and easy.

dpgeorge, 2017-10-04:
1) is implemented in dfa563c71fead671931e91b786cd649eed517f92; it actually costs a few bytes on x86 archs (but not others), no idea why.

And in 23faf88cab10a216eb24a30b4f33edd17183522c I fixed (3) and made that null check only enabled when debugging (asserts) are enabled, which saves a bit of code space.

dpgeorge, 2017-10-04:
> from a perspective of extmod development, it was surprising to discover that mp_buffer_info_t.buf can also be NULL (event though it's documented in the code). Should we also expect to receive 0-terminated str here?

No, the buffer protocol is all about accessing a chunk of memory associated with some object's data, and it may be a subset of the data (eg memoryview usage).  So you should never access more than `buf.len` bytes of data.

closed 2017-10-04 by dpgeorge