micropython/micropython #3338: cannot get json request with usocket

ashareye, 2017-09-29:
hi

   I use esp8266 as webserver with usocket, when posted a request as follow:

> POST /polling HTTP/1.1
Content-Type: application/json; charset=utf-8
Host: 192.168.0.199
Connection: close
Content-Length: 34
> 
> {"dryin1":"true","dryin2":"false"}

 usocket could only get http header and stopped on b'/r/n', it could not get the last line {"dryin1":"true","dryin2":"false"}，and if I post  request again, it would get the last line before the new request, why? here is my program:
> def starts():
....import usocket,ujson
....s = usocket.socket()
....s.setsockopt(usocket.SOL_SOCKET, usocket.SO_REUSEADDR, 1)
....s.bind(usocket.getaddrinfo('0.0.0.0', 80)[0][-1])
....s.listen(1)
....while True:
........cl, addr = s.accept()
........method, path, dummy = cl.readline().split()
........print(addr, method, path)
........for i in range(0, 20):   # only 7
............line = cl.readline()
............print(line)  # stop printing with b'\r\n' and will print last json line at next request

please give me some advice, thanks!

dpgeorge, 2017-10-03:
The `readline()` method will not return until it gets a newline character, and probably the end of your json data doesn't have a newline.  Therefore it waits forever until the socket closes.  To handle data in a POST request you should stop reading lines at the empty line, interpret the Content-Length header, then read that many bytes.

In general these kinds of questions are best asked on the furum at https://forum.micropython.org

ashareye, 2017-10-03:
Thank you for your reply, but it might not seem so because I also used "cl.read()" and got the same result. Now I have to use the query instead of json posting and it works well. I'll try to solve this problem again when I have free time. 
What makes me curious is that when esp8266 requested to the server, then the json response of server was resolved correctly, and the server used the same function. 

thanks again

closed 2017-10-21 by pfalcon